enjoy-digital/litex #1858: WSL2 Debian Gowin Toolchain Path

mlg556, 2023-12-23:
on wsl2 Debian under windows 11, even though the Gowin executable "gw_sh" is under my $PATH, litex can't find it. the issue seems to come from [gowin.py#L168-L169](https://github.com/enjoy-digital/litex/blob/master/litex/build/gowin/gowin.py#L168-L169):

```
if os.uname().release.find("
WSL") > 0:
    gw_sh += ".exe"
```

I don't understand why an executable under wsl debian would have an .exe extension. and my system doesn't, maybe others do? If I delete this condition which appends ".exe" to "gw_sh", litex manages to successfully build my soc. maybe we should just remove this condition? I can make a pull request, but I first probably have to understand why it's there in the first place :)

[image]

exitrip, 2024-04-17:
Conversely, I am using WSL2 for Litex builds and have Gowin IDE installed in Windows natively, with my PATH pointing to the native bin folder.  The EXE check is needed for my setup to work.

linhz0hz, 2024-04-21:
> Conversely, I am using WSL2 for Litex builds and have Gowin IDE installed in Windows natively, with my PATH pointing to the native bin folder. The EXE check is needed for my setup to work.

Does WSL2 allow interfacing with windows installed Gowin? I tried this and it gets 
Cannot open Verilog file 'C:\mnt\c\fpga\litex\litex_demo\build\top.v'

kamejoko80, 2024-05-26:
Export PATH environment variable below:

export PATH=$PATH:/mnt/c/Gowin/Gowin_V1.9.9.03_x64/IDE/bin

It works for me.

stek29, 2024-06-03:
I just symlinked gw_sh.exe to gw_sh, thanks for finding this, took me a while to realize why it won't see my toolchain.

```sh
cd IDE/bin
ln -s gw_sh gw_sh.exe
```

would it be acceptable to have some way to adjust this? what if it tries to look for gw_sh on WSL2 too, and only looks for gw_sh.exe if gw_sh isn't found?
or only look for gw_sh.exe on wsl1 and not wsl2

@exitrip are you using wsl2 or wsl1?

exitrip, 2024-06-04:
Wal!