enjoy-digital/litex #1860: Html doc not generating for TUL PYNQ Z2

ZeroX29a, 2023-12-27:
I have tried ./tul_pynq_z2.py --build --doc and i get this error and there is no html files in the doc/_build directory
```
INFO: [Designutils 20-2272] Running write_bitstream with 8 threads.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Writing bitstream ./tul_pynq_z2.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-1876] WebTalk data collection is mandatory when using a ULT device. To see the specific WebTalk data collected for your design, open the usage_statistics_webtalk.html or usage_statistics_webtalk.xml file in the implementation directory.                                                                                                  
INFO: [Common 17-83] Releasing license: Implementation
9 Infos, 14 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:10 ; elapsed = 00:00:08 . Memory (MB): peak = 3090.105 ; gain = 5.242 ; free physical = 5045 ; free virtual = 9698
# quit
INFO: [Common 17-206] Exiting Vivado at Thu Dec 28 03:03:07 2023...
Generate the documentation by running `sphinx-build -M html /home/litex/litex_installation/litex-boards/litex_boards/targets/build/tul_pynq_z2/doc/ /home/litex/litex_installation/litex-boards/litex_boards/targets/build/tul_pynq_z2/doc/_build`
identifier_mem@f0000800: Found memory that's 8 x 41 (but memories aren't documented yet)
sh: 1: sphinx-build: not found```
seems like documentation is WIP, Is there any way to get board io key and id values for pynq z2 (Im a beginner in migen and litex, and intermediate in VHDL )

AndrewD, 2023-12-27:
> sh: 1: sphinx-build: not found

^^^ have you installed sphinx? This is used to generate the documentation

ZeroX29a, 2023-12-28:
Installing sphinx helped with the issue, initially I thought installing Litex full and oss toolchain must have had this tool. My mistake.

closed 2023-12-28 by ZeroX29a