enjoy-digital/litex #1866: No serial output on tang 9k when selecting other sys-clk-freq

rhgndf, 2024-01-03:
Running `python -m litex_boards.targets.sipeed_tang_nano_9k --build --flash` works as expected with the litex bios output seen, but running:
- `python -m litex_boards.targets.sipeed_tang_nano_9k --build --flash --sys-clk-freq 30e6`
- `python -m litex_boards.targets.sipeed_tang_nano_9k --build --flash --sys-clk-freq 50e6`
- `python -m litex_boards.targets.sipeed_tang_nano_9k --build --flash --sys-clk-freq 54e6`

All do not produce any output from the litex bios. The report generated do not seem to have any timing or utilitization issues.

luxuan12512, 2024-01-05:
hello， you can annotation or del the code
        kwargs["integrated_rom_size"] = 0
and

self.bus.add_region("rom", SoCRegion(
            origin = self.bus.regions["spiflash"].origin + bios_flash_offset,
            size   = 64*kB,
            linker = True)
        )
        self.cpu.set_reset_address(self.bus.regions["rom"].origin)

rhgndf, 2024-01-05:
Hi, I have tried the above and it still doesn't work. Although, finding a solution that doesn't require removing the spiflash XIP code would be nice to free up some BRAMs for other uses.

AndrewD, 2024-01-05:
The default --sys-clk-freq is 27mhz which matches the input clock, so if changing --sys-clk-freq doesn't work then then may be an issue with the PLL setup in litex. Are there any warnings from the toolchain?

If you have an oscilloscope you check if the design is working at all by looking for serial output and checking the baud rate.

FYI disabling XIP as per the comment by @luxuan12512 is unlikely to help.

rhgndf, 2024-01-06:
One thing to note is that the LED chaser works as expected. However, the UART pins seem to be internally routed from the FPGA to the usb programmer chip. Changing baud rates randomly also does not seem to even get garbage outputted. I guess I could try to change the fpga pin assignment to something that is broken out.

There are a lot of warnings but those does not seem related:

```
Analyzing Verilog file '/.../tang9k/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v'
Analyzing Verilog file '/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v'
Compiling module 'sipeed_tang_nano_9k'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":21)
WARN  (EX3780) : Using initial value of 'builder_interface0_err' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":288)
WARN  (EX3780) : Using initial value of 'main_basesoc_basesoc_adr_burst' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":364)
WARN  (EX3780) : Using initial value of 'main_basesoc_basesoc_ram_bus_err' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":373)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespimmap' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":420)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespimmap_bus_err' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":434)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespimmap_dummy' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":441)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespimmap_source_first' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":448)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore0' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":459)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore1' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":460)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore2' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":461)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore3' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":462)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore_en_int' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":474)
WARN  (EX3780) : Using initial value of 'main_basesoc_litespisdrphycore_source_first' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":490)
WARN  (EX3780) : Using initial value of 'main_basesoc_ram_adr_burst' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":509)
WARN  (EX3780) : Using initial value of 'main_basesoc_ram_bus_ram_bus_err' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":517)
WARN  (EX3780) : Using initial value of 'main_basesoc_rx_source_first' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":538)
WARN  (EX3780) : Using initial value of 'main_basesoc_rx_source_last' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":539)
WARN  (EX3780) : Using initial value of 'main_basesoc_uart_rx_fifo_replace' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":633)
WARN  (EX3780) : Using initial value of 'main_basesoc_uart_tx_fifo_replace' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":691)
WARN  (EX3780) : Using initial value of 'main_basesoc_uart_tx_fifo_sink_first' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":692)
WARN  (EX3780) : Using initial value of 'main_basesoc_uart_tx_fifo_sink_last' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":693)
WARN  (EX3780) : Using initial value of 'main_basesoc_vexriscv' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":745)
WARN  (EX3780) : Using initial value of 'main_hyperram_bus_err' since it is never assigned("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":761)
Extracting RAM for identifier 'rom'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":2454)
Extracting RAM for identifier 'sram'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":2469)
Extracting RAM for identifier 'mem'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":2492)
Extracting RAM for identifier 'storage'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":2508)
Extracting RAM for identifier 'storage_1'("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":2529)
[... cut ...]
WARN  (EX0211) : The output port "io_cpu_writesPending" of module "DataCache" has no driver, assigning undriven bits to Z, simulation mismatch possible("/.../tang9k/pythondata-cpu-vexriscv/pythondata_cpu_vexriscv/verilog/VexRiscv.v":5216)
[5%] Running netlist conversion ...
WARN  (CV0021) : Inout IO_psram_dq[15:8] is unused("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":22)
WARN  (CV0019) : Inout IO_psram_rwds[1] is unused("/.../tang9k/src/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.v":23)
```

rhgndf, 2024-01-11:
Did a few more tests, decreasing the frequency works, with output on the uart, but increasing frequency does not.

trabucayre, 2024-01-11:
Hi,
When you try to increase clock frequency and you observe this type of behavior it's required to check reports/output from toolchain.
More precisely max frequency is shown in build/sipeed_tang_nano_9k/gateware/impl/pnr/project.tr.html.
And for this board max frequency is around 35/40MHz.
50MHz fails, 35MHz works for me

rhgndf, 2024-01-11:
I think I was checking the synthesis results instead of the pnr results. It works now when the pnr passes.

[image]

rhgndf, 2024-01-11:
Hmm, not quite. At 42MHz the pnr timing passes but there is no output.

[image]

rhgndf, 2024-01-11:
I think this may be a toolchain problem? I'm not sure. There is uart output for the standard cpu configuration but once passing --cpu-variant full the problem above appears. Both of them pass timing verification in the pnr stage, so it's probably not a problem with litex?

enjoy-digital, 2024-01-12:
@rhgndf: It would be interesting to output a divided version of the clock to one of the pin of the board and observe it with a scope and do the same for the UART TX. If correct, this would exclude PLL configuration issues and if not, we would need to check the PLL.

rhgndf, 2024-01-12:
The board has an LED chaser, which is probably close enough? The LEDs light up at approximately the correct timing as far as I can tell.

It seems like it might be data corruption somewhere since in the failed attempts, it outputs BIOS CRC Error.