enjoy-digital/litex #1869: hello,i have quetion about it . i dele the del the code  kwargs["integrated_rom_size"] = 0  and  self.bus.add_region("rom", SoCRegion( origin = self.bus.regions["spiflash"].origin + bios_flash_offset, size = 64*kB, linker = True) ) self.cpu.set_reset_address(self.bus.regions["rom"].origin)  and run in terminal: ./sipeed_tang_nano_9k.py --build is no problem,and uart is out but run in termimal: ./sipeed_tang_nano_9k.py --with-spi-sdcard --build have error ERROR (RP0002) : The number(31) of BSRAM in the design exceeds the resource limit(26) of current device. And RAM_STYLE maybe the useful user assignment to change the inference result  if i don't del the code to run : ./sipeed_tang_nano_9k.py --with-spi-sdcard --build is no error but uart isn't out  So what should I do to be able to have urat output in the --with-spi-sdcard thanks!!!

luxuan12512, 2024-01-05:


luxuan12512, 2024-01-05:
hello,i have quetion about it .
i dele the del the code

kwargs["integrated_rom_size"] = 0

and

self.bus.add_region("rom", SoCRegion(
origin = self.bus.regions["spiflash"].origin + bios_flash_offset,
size = 64*kB,
linker = True)
)
self.cpu.set_reset_address(self.bus.regions["rom"].origin)

and run in terminal:
./sipeed_tang_nano_9k.py --build
is no problem,and uart is out
but run in termimal:
./sipeed_tang_nano_9k.py --with-spi-sdcard --build
have error
ERROR (RP0002) : The number(31) of BSRAM in the design exceeds the resource limit(26) of current device. And RAM_STYLE maybe the useful user assignment to change the inference result

if i don't del the code to run :
./sipeed_tang_nano_9k.py --with-spi-sdcard --build
is no error but uart isn't out

So what should I do to be able to have urat output in the --with-spi-sdcard
thanks!!!

enjoy-digital, 2024-01-07:
Hi @luxuan12512,

I'll have a look soon, for now I just make the issue a bit more easy to undertand through ChatGPT:
[image]

trabucayre, 2024-01-09:
I must admit your question is not really clear but:
- this FPGA has not a big amount of BRAM. With default configuration `rom` region may uses internal RAM, but requires, more or less, all of them.
- the consequence: when you try to add spi-scard support, this driver requires RAM too and now there are not enough RAM available
- this why `rom` is located in spi flash instead of RAM

For the second part of this issue (no UART): have you flashed `bios.bin` into the flash? Default gateware try to read CPU firmware in this storage, so if the flash is blank or with a different content it may make sense you have nothing executed.

I have also noticed an issue with the UART interface, first time after pluged, nothing is displayed: I have to close my terminal emulator and reopen it to obtain a prompt.

I hope having answered to your issue.

blark, 2024-02-14:
I am having the same issue here. I used:

`./sipeed_tang_nano_9k.py --prog-kit openfpgaloader --flash --build` 

I'm on an M1 and running the build from UTM (QEMU), which works fine but the flash hangs after:

```
Jtag frequency : requested 6.00MHz   -> real 6.00MHz
Parse file Parse /usr/local/share/litex/litex/litex-boards/litex_boards/targets/build/sipeed_tang_nano_9k/gateware/sipeed_tang_nano_9k.fs:
Done
DONE
Jtag frequency : requested 2.50MHz   -> real 2.00MHz
erase SRAM Done
erase Flash Done
```

When I copy the `.fs` to my Mac and flash it with openFPGAloader, the led chaser works fine but I don't get any response from the serial port. Is there some other step of copying over the bios that I'm missing?