enjoy-digital/litex #1873: Clock domain crossing for peripherals

rhgndf, 2024-01-11:
Is it possible to use bus.add_slave to add peripherals on a different clock domain? Possible use cases are running the spi flash or sd card at a higher frequency than the main cpu to allow for faster accesses.

enjoy-digital, 2024-01-12:
Hi  @rhgndf,

for now this is not directly integrated in the framework and still have to be done manually in the user logic (ie first add a CDC to run the MMAP interface in `sys_clk`).  This is a features that will probably be added in the future.

rhgndf, 2024-01-12:
Thanks for the reply.

If I'm not wrong to do this manually this involves making a new Wishbone/AXI CDC wrapper where all the bus lines are used in a BusSynchronizer for both directions?

enjoy-digital, 2024-01-12:
For Wishbone, we don't have yet a CDC module, but for AXI-Lite, you can use: https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/axi/axi_lite.py#L489-L527