enjoy-digital/litex #1875: vexriscv_smp: only 32 CSR locations supported with current mem_mam

AndrewD, 2024-01-19:
The current memory map allows for 32 CSR locations:
`INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).`

This can be expanded to 64 by specifying `--csr-address-width=15`, however the cpu mem_map places clint straight after csr
[            "csr":      0xf000_0000,
            "clint":    0xf001_0000,](https://github.com/enjoy-digital/litex/blob/b19d992f231699f1ac7d07b73a15deb400d23b59/litex/soc/cores/cpu/vexriscv_smp/core.py#L149-L150)

So this causes the following build error:
```
ERROR:SoCBusHandler:Region overlap between clint and csr:
ERROR:SoCBusHandler:Origin: 0xf0010000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
ERROR:SoCBusHandler:Origin: 0xf0000000, Size: 0x00020000, Mode: RW, Cached: False Linker: False
```

Can the mem_map be adjusted easily to support additional CSR locations?

trabucayre, 2024-01-22:
Hi,
True: CSR region is a bit small.

With https://github.com/enjoy-digital/litex/pull/1876 it is possible to move CSR/CLINT/PLIC (and consequently it's possible to increase CSR region).

This may be done by overriding region directly from the target and/or by passing region base address using CLI args.

Is this solution is good to you?

AndrewD, 2024-01-24:
Thanks - that should address the issue.

enjoy-digital, 2024-02-05:
As noted by @AndrewD, https://github.com/enjoy-digital/litex/pull/1885 can also be useful for this. You now have the possibility to move `CLINT`/`PLIC` or reduce CSR paging.

closed 2024-03-11 by AndrewD