enjoy-digital/litex #1882: NuttX + LiteEth: Potential regression when using Hybrid Etherbone/Ethernet.

enjoy-digital, 2024-02-01:
Reported by @AndrewD, to look at this:
- Run NuttX with LiteEth on a board with `add_ethernet `.
- Run NuttX with LiteEth on a board with `add_etherbone  + hybrid mode`.
- If issue with the latter, fix. 

Note that it's possible the issue with NuttX can also be reproduced with Linux if easier to test.

AndrewD, 2024-02-01:
This is the way we integrated Hybrid Etherbone/Ethernet for testing on Arty:

```
        # Ethernet and/or Etherbone ---------------------------------------------------------------------
        if ethernet or etherbone:
            self.ethphy = LiteEthPHYMII(
                clock_pads = self.platform.request("eth_clocks"),
                pads       = self.platform.request("eth"))
            # Etherbone with optional Ethernet
            if etherbone:
                self.add_etherbone(
                    phy=self.ethphy,
                    ip_address=etherbone_ip,
                    with_ethmac=ethernet,
                )
            # Ethernet only
            elif ethernet:
                self.add_ethernet(phy=self.ethphy)
```

enjoy-digital, 2024-02-05:
Thanks @AndrewD, @trabucayre has been able to reproduce the issue (even with LiteX netboot). This is related to the integration improvements. We just need to do a few iterations to find the root cause and we'll fix it.

trabucayre, 2024-02-07:
Hi @AndrewD . I have fixed a small issue in `add_etherbone` method (https://github.com/enjoy-digital/litex/commit/13c57e8304d8c685268be67400a66a5f7c08c4a2).
But you have to provides a *mac_address* parameter different than default value to avoid conflict with ethernet part.

Tested on arty with nuttx (master branch).

AndrewD, 2024-02-07:
Excellent, I'll integrate this tomorrow.

I'd experimented with a separate mac address without success, however the clocking fix looks like the missing piece.

enjoy-digital, 2024-02-07:
Hi @AndrewD,

we are going to add parameters to specify the different IP/MAC addresses and also add some check to ensure there are no collisions between Etherbone and Ethernet. @trabucayre will do the changes this morning and provide an update.

trabucayre, 2024-02-07:
Hi,

See this [PR](https://github.com/enjoy-digital/litex/pull/1886).
With these modifications is possible to pass all configurations (etherbone IP/MAC + ethernet local/remote IP and MAC).

AndrewD, 2024-02-21:
Thanks @trabucayre and @enjoy-digital, I've confirmed this is now working nicely with bios and nuttx.

enjoy-digital, 2024-02-21:
Good, thanks @AndrewD for the feedback.