enjoy-digital/litex #1883: Allow CSR paging 0x400

smunaut, 2024-02-02:
Is there a particular reason that CSR paging is 0x800 minimum ?

0x400 seems to build at least (if I allow it) and to me it looks like 256 registers in a bank is quite a bit already (never hit that limit myself, nowhere even close to it).

Of course I know you need to set the limit somewhere, but recently several new blocks have popped up in a default SoC (like sdram has a couple more CSR banks, PCIe endpoint now has one), which tends to already fill a bare default SoC max 32 locations with quite a few things.

And combined with the fact that  youcan only force / control the first layer of blocks address (with csr_map), you often want your stuff to be at that level and thus occupy one slot and you can quickly get to 32.

AndrewD, 2024-02-02:
Related to #1875

enjoy-digital, 2024-02-05:
Thanks @smunaut for #1885. To ease readability and eventually extend it in the future, I just made the values explicit with  https://github.com/enjoy-digital/litex/commit/57bc0369c728c3666f51c3da9281c122132ee594.

closed 2024-02-05 by enjoy-digital