enjoy-digital/litex #1891: new Spartan6 programmer

suarezvictor, 2024-02-17:
I find this code to program Spartan6 devices useful. It's based on OpenOCD, and it works.
Can be added here: https://github.com/enjoy-digital/litex/blob/master/litex/build/xilinx/programmer.py
Any interest?

```
# Copyright (C) 2024 Victor Suarez Rovere <suarezvictor@gmail.com>
# License: BSD 2-Clause License

from litex.build.generic_programmer import GenericProgrammer
class Spartan6Programmer(GenericProgrammer):
    needs_bitreverse = False

    def __init__(self, config): 
        GenericProgrammer.__init__(self)
        self.config = config #openocd script (.cfg) like those found on /usr/share/openocd/scripts/board

    def load_bitstream(self, bitstream):
        script = "; ".join([
            "init",
            "xc6s_program xc6s.tap",
            "pld load 0 {{{}}}".format(bitstream),
            "exit",
        ])
        self.call(["openocd", "-f", self.config, "-c", script])

```

enjoy-digital, 2024-06-13:
Hi @suarezvictor,

this is interesting but a bit different from the current approach where we have the OpenOCD configuration files in litex_boards/prog: https://github.com/litex-hub/litex-boards/tree/master/litex_boards/prog

If spartan6 programming is currently not supported through OpenOCD in LiteX, we could add a configuration file in litex_board/prog and in the long term we could try to see if generating the configuration file for all the FPGA on the fly could be interesting and then, why not removing the configuration files in LiteX Boards.

suarezvictor, 2024-06-13:
I should test that approach, how to select a custom config file? since the pins and bitmasks for XC6 + FTDI chip varies from board to board