enjoy-digital/litex #1914: litex_sim of CV32E41P is stuck

pcotret, 2024-04-02:
Installed LiteX on a new VM.
```bash
 litex_sim --cpu-type=cv32e41p
```
Stalls:
```bash
[...]
g++    veril.o sim_init.o verilated.o verilated_vcd_c.o Vsim__ALL.a   modules.o pads.o sim.o libdylib.o parse.o -lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent  -o Vsim -lm -lstdc++ 
make[1]: Leaving directory '/opt/litex_install/build/sim/gateware/obj_dir'
make: Leaving directory '/opt/litex_install/build/sim/gateware'

[serial2tcp] loaded (0x55ea5a37c190)
[serial2console] loaded (0x55ea5a37c190)
[jtagremote] loaded (0x55ea5a37c190)
[xgmii_ethernet] loaded (0x55ea5a37c190)
[spdeeprom] loaded (addr = 0x0)
[gmii_ethernet] loaded (0x55ea5a37c190)
[clocker] loaded
[ethernet] loaded (0x55ea5a37c190)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0
```
## Setup
- Python 3.8.10
- Ubuntu 20.04.6

mohamedElbouazzati, 2024-04-08:
@pcotret 

This commit of pythondata-cpu-cv32e41p should work : 
```
commit 339d2f23acd62f6c003333e062133db987413360 (HEAD)
Merge: 48516c9 cf0ab06
Author: LiteX Robot <robot@timvideos.us>
Date:   Thu Feb 17 12:41:00 2022 +0000
    Merge commit 'cf0ab06e28a7845d0ce736b44a27d1bfab47074a'
```

```
[spdeeprom] loaded (addr = 0x0)
[serial2tcp] loaded (0x55b87941d190)
[serial2console] loaded (0x55b87941d190)
[clocker] loaded
[xgmii_ethernet] loaded (0x55b87941d190)
[ethernet] loaded (0x55b87941d190)
[jtagremote] loaded (0x55b87941d190)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr  8 2024 17:06:22
 BIOS CRC passed (dde02867)

 LiteX git sha1: d3ea9123

--=============== SoC ==================--
CPU:		CV32E41P @ 1MHz
BUS:		wishbone 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> help

LiteX BIOS, available commands:

flush_cpu_dcache         - Flush CPU data cache
crc                      - Compute CRC32 of a part of the address space
ident                    - Identifier of the system
help                     - Print this help

serialboot               - Boot from Serial (SFL)
reboot                   - Reboot
boot                     - Boot from Memory

mem_cmp                  - Compare memory content
mem_speed                - Test memory speed
mem_test                 - Test memory access
mem_copy                 - Copy address space
mem_write                - Write address space
mem_read                 - Read address space
mem_list                 - List available memory regions

litex> 

```

pcotret, 2024-04-08:
Yep, thanks @mohamedElbouazzati, it works.
I'll leave this issue open. I'll try to search what broke the execution with a newer `pythondata-cpu-cv32e41p`.