enjoy-digital/litex #1922: Can't boot linux with MAIN_RAM sizes above 512mb

JoyBed, 2024-04-10:
I stumbled upon an interesting bug. Linux is unable to boot if the memory size is above 512mb. In litex BIOS the whole ram passes the tests and its working but as linux start to boot it cant boot if I have ram size specified above 512mb. Where does this limitation come from? 
[image]

JoyBed, 2024-04-20:
Well... When I limit it in the dts to 512mb then the linux can boot but the biggest problem I always had is NaxRiscv. The main_ram works on ANY softcore but not on the NaxRiscv, here is the screenshot:
[image]
Thats when its connected to main ram thru peripheral bus, when I connect the main ram directly to the axi4 ports it gets stuck at the "memtest at 0x40000000" and doesnt even count.

Dolu1990, 2024-04-22:
Hi,

I just tested via : 
litex_sim --cpu-type=naxriscv --with-sdram   --sdram-module=MT41K128M16 --sdram-data-width=64 

I got :
```
--=============== SoC ==================--
CPU:		NaxRiscv 32-bit @ 1MHz
BUS:		wishbone 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		1.0GiB 64-bit @ 8MT/s (CL-6 CWL-5)
MAIN-RAM:	1.0GiB

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (8.0KiB)...
  Write: 0x40000000-0x40002000 8.0KiB   
   Read: 0x40000000-0x40002000 8.0KiB   
Memtest OK
Memspeed at 0x40000000 (Sequential, 8.0KiB)...
  Write speed: 2.9MiB/s
   Read speed: 3.4MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Cancelled

--============= Console ================--
litex> mem_test 0x40000000 0x1000
Memtest at 0x40000000 (4.0KiB)...
  Write: 0x40000000-0x40001000 4.0KiB   
   Read: 0x40000000-0x40001000 4.0KiB   
Memtest OK

litex> mem_test 0x50000000 0x1000
Memtest at 0x50000000 (4.0KiB)...
  Write: 0x50000000-0x50001000 4.0KiB   
   Read: 0x50000000-0x50001000 4.0KiB   
Memtest OK

litex> mem_test 0x60000000 0x1000
Memtest at 0x60000000 (4.0KiB)...
  Write: 0x60000000-0x60001000 4.0KiB   
   Read: 0x60000000-0x60001000 4.0KiB   
Memtest OK

litex> mem_test 0x70000000 0x1000
Memtest at 0x70000000 (4.0KiB)...
  Write: 0x70000000-0x70001000 4.0KiB   
   Read: 0x70000000-0x70001000 4.0KiB   
Memtest OK
```

What command line did you used ?
One specific thing about NaxRiscv, is that the CPU is generated with the exact knowledge of "where is some ram i can access".
So if there is a bug there it could create your case.

JoyBed, 2024-04-22:
I used this:
./xilinx_zybo_z7_20.py --variant=original --cpu-type=naxriscv --xlen=64 --scala-args='rvc=true,rvf=true,rvd=true,alu-count=2,decode-count=2,mmu=true' --with-fpu --with-rvc --with-ps7 --bus-standard=axi-lite --with-spi-sdcard --sys-clk-freq=125e6 --with-xadc --csr-json zybo.json --uart-baudrate=2000000 --build --update-repo=wipe+recommended --vivado-synth-directive=PerformanceOptimized --vivado-route-directive=AggressiveExplore --with-hdmi-video-framebuffer --l2-bytes=262144 --l2-ways=16 --with-jtag-tap
When I generate a NaxRiscv with mbus thats connected to DRAM than it freezes at mem_test at 0x40000000, when I generate it with DRAM connected to pbus than I have those data errors as in the screenshot.

Dolu1990, 2024-04-22:
> when I generate it with DRAM connected to pbus

Hmm should realy not do that, the Nax SoC is realy intended to use cacheable memory through mbus. Things through pbus may not support atomic access and stuff like that.

> When I generate a NaxRiscv with mbus thats connected to DRAM than it freezes at mem_test at 0x40000000,

Ahhh that is one thing. Probably related to the zynq nature of the FPGA. Not sure if there is a way to setup a simulation of the SoC / zynq with vivado ?

Dolu1990, 2024-04-22:
Maybe the reason why this isn't working, is that if let's say, we specify that the memory is on the AXI bus at 0x40000000, then  naxriscv can access it, but the memory accesses on that axi bus will be emited without that 0x40000000 offset.

I don't know what is the expected behaviour from the zynq / litex side

JoyBed, 2024-04-22:
> Maybe the reason why this isn't working, is that if let's say, we specify that the memory is on the AXI bus at 0x40000000, then  naxriscv can access it, but the memory accesses on that axi bus will [...]

I dont quite understand now. If it has it on the mbus then accesses on the mbus are done without the 0x40000000 offset? So a access to 0x45000000 thru mbus emits address 0x05000000 ?

Dolu1990, 2024-04-22:
> So a access to 0x45000000 thru mbus emits address 0x05000000 ?

Yes, i need to double check but that is quite possible.

JoyBed, 2024-04-23:
> > So a access to 0x45000000 thru mbus emits address 0x05000000 ?
> 
> 
> 
> Yes, i need to double check but that is quite possible.

I tried it with mbus connected to DRAM with bus address offset stripping (the 0x40000000) still when DRAM is connected to the mbus it locks up at the bootup mem_test. I dont know whats wrong, before experimenting with NaxRiscv I was using Rocket and that had its memory bus connected straight to the DRAM and it worked like a charm. Before that I used VexRiscv which had the same main_ram address 0x40000000 as NaxRiscv and it was working too. Even the Microwatt and Serv were able to work with it. NaxRiscv is the only one doing this. Im out of ideas.

Dolu1990, 2024-04-23:
I'm looking at it. Trying to get the offset to be preserved.

Also, did you tried vexriscv_smp cpu ?

JoyBed, 2024-04-23:
> I'm looking at it. Trying to get the offset to be preserved.
> 
> 
> 
> Also, did you tried vexriscv_smp cpu ?

Yes, and it was working and booting linux just fine, but not with memory bus to dram as the memory bus of the vexriscv smp has litedram interface not axi4 as the PS7 of the zynq has. I also successfully booted linux on Rocket and Microwatt and I wanted to move to NaxRiscv for performance reasons, also I can fit 2 NaxRiscv cores into my FPGA and with Linux variant of Rocket I can only fit one.

JoyBed, 2024-04-24:
Very strange behaviour, thru mbus it locks up at mem_test evne when i specified no L2 cache. I also tried it again with dram to pbus without L2 to see if it makes a difference. Still the same behaviour.
[image]

JoyBed, 2024-04-25:
Also if the DRAM is connected thru pbus and I just bypass the non functional addresses then it can even boot:
[image]
Even linux booted when I specified in device tree that those addresses are reserved memory regions.

Dolu1990, 2024-04-25:
> Also if the DRAM is connected thru pbus and I just bypass the non functional addresses then it can even boot:

I do not expect it to boot in that configuration. Also performances will be very bad.

I pushed a potential fix with : 
https://github.com/enjoy-digital/litex/pull/1940

With this one, mBus accesses will preserve the full 32 bits address, instead of removing the 0x40000000 offset.

I don't have any zynq board, let's me know how it goes :)

Also, keep in mind, VexiiRiscv is very close to get feature parity with performance not too far away. (WIP)

JoyBed, 2024-04-25:
> I do not expect it to boot in that configuration. Also performances will be very bad.
>
> I pushed a potential fix with : #1940
>
> With this one, mBus accesses will preserve the full 32 bits address, [...]

I tried it but still locked up on mem_test when DRAM is connected to mbus:
[image]
I dont understand this behaviour, its very strange.

Dolu1990, 2024-04-25:
Did you checked it passes the timings ?

Dolu1990, 2024-04-25:
Weird, I just tested on Digilent nexys video, can run debian just fine
Did you deleted the pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/NaxRiscvLitex_????.v files before retrying with the fixes ?
Else it will not regenerate the naxriscv SoC and use it as a cache.

JoyBed, 2024-04-25:
Yes, I deleted the generated verilog files. I dont understand this strange behavior at all. My goal is to boot Debian/Fedora on it as I did with Rocket. I am thinking maybe adding a litescope to or jtagbone to the SoC to see the signals on mbus<->dram bus to see what is going on.

Dolu1990, 2024-04-25:
>  to see the signals on mbus<->dram bus to see what is going on.

Yes that would be a the way to proceed.
probing the mbus, aswell as probing the dbus which get out of the CPU itself

Idealy, instead of relying on hardware debug, we would run a simulation, that would give use full visibility on what is happening.

JoyBed, 2024-04-25:
> Yes that would be a the way to proceed. probing the mbus, aswell as probing the dbus which get out of the CPU itself
>
> Idealy, instead of relying on hardware debug, we would run a simulation, that [...]

I dont know if simulation would do anything as the mbus is connected to the PS7 block which contains hardened components not softcores. Also you cant connect to UART in a Vivado simulation.

JoyBed, 2024-04-28:
Here it is with Rocket softcore. Memory bus of the Rocket connected straight to DRAM. Working as with any other softcore except the NaxRiscv.
[image]

Dolu1990, 2024-04-28:
can you send your custom board files ? for me to recreate. Maybe it is the memory region definition which is messed up.

JoyBed, 2024-04-28:
> can you send your custom board files ? for me to recreate. Maybe it is the memory region definition which is messed up.

target and platform files ?

Dolu1990, 2024-04-28:
yes

JoyBed, 2024-04-28:
> yes

Here you go.
[custom_zybo.zip](https://github.com/enjoy-digital/litex/files/15141485/custom_zybo.zip)
There is the platform and target file and also the modified Zynq7000 core file so that the HP ports have the ACLK taken from softcore bus, otherwise it will not work.
Also use this pull request, im using that function: [https://github.com/enjoy-digital/litex/pull/1522](url)

Dolu1990, 2024-04-30:
Just tried now.

```
[info] LitexMemoryRegion(SM(0x80000000, 0x80000000),io,p)
[info] LitexMemoryRegion(SM(0x0, 0x20000),rxc,p)
[info] LitexMemoryRegion(SM(0x10000000, 0x2000),rwxc,p)
[info] LitexMemoryRegion(SM(0x40000000, 0x40000000),rwxc,p)
[info] LitexMemoryRegion(SM(0xf0000000, 0x10000),rw,p)
```

0x40000000, 0x40000000   => address, size

which would normaly mean that there should be 1 GB accessible at 0x40000000

The thing is, "[info] LitexMemoryRegion(SM(0x80000000, 0x80000000),io,p)" is overlapping that memory range.
You need to push that memory region (0x80000000, 0x80000000) to (0xC0000000, 0x40000000)

note it should not be on ",p)"   but on ,"m)" else you will get bad performances (meaning not pbus, but mbus)

io_regions           = {0x4000_0000: 0xbc00_0000} # Origin, Length.
Does it mean that the DDR is mapped as if it was a io region ?

JoyBed, 2024-04-30:
Uncomment these lines:
'''python

#if hasattr(self.cpu, "add_memory_buses"):
    #self.cpu.add_memory_buses(address_width = 32, data_width = 64)
'''
That would connect the DRAM to mbus instead of pbus. Also you can change the parameter sdram_size to half so ut doesnt overlap with anything(if i understood correctly).

Dolu1990, 2024-04-30:
> The thing is, "[info] LitexMemoryRegion(SM(0x80000000, 0x80000000),io,p)" is overlapping that memory range.

Ahhh foget that, my bad XD

So, the only way to debug i would say would be with the logic analyser.

Also one thing, with NaxRiscv you realy need to use the argument : --bus-standard axi-lite 
(the axi-lite <> wishbone) is bugged i think. That may explaine the crashes you had before. (not talking of the memory range)

JoyBed, 2024-04-30:
> Ahhh foget that, my bad XD
>
>
>
> So, the only way to debug i would say would be with the logic analyser.
>
>
>
> Also one thing, with NaxRiscv you realy need to use the argument : --bus-standard axi-lite [...]

Yeah. I know that, I have the pbus as axi-lite and the mbus to dram is full axi4. I dont usually use wishbone when not specifically needed. I try to always use the bus standard the core has as its native output.

Dolu1990, 2024-04-30:
Hmm, so yes, at this point realy need to add probes / logic analyser, and trace memory access at 0x60000000 to see until which point they reach.
Also, note that today i also got VexiiRiscv to run debian, there is one hardware bug i know of which trigger instruction access fault on some specific binaries, but soon it should be fixed.

JoyBed, 2024-04-30:
> Hmm, so yes, at this point realy need to add probes / logic analyser, and trace memory access at 0x60000000 to see until which point they reach.
>
> Also, note that today i also got VexiiRiscv to run [...]

Wait what. VexRiscv running Debian? Distros require RV64GC and VexRiscv is RV32GC or did I missed something?

Dolu1990, 2024-04-30:
> did I missed something?

Yes, you missed the "ii"
VexiiRiscv, not VexRiscv ^^

JoyBed, 2024-04-30:
> > did I missed something?
> 
> 
> 
> Yes, you missed the "ii"
> 
> VexiiRiscv, not VexRiscv ^^

Ooooh, now I checked it out and it looks cool. When it gets integrated into Litex?  How did you got it to the FPGA board? Im totaly new and lost in Spinal/Scala.

Dolu1990, 2024-04-30:
There is a PR which is WIP and not up to date : 
https://github.com/enjoy-digital/litex/pull/1923

I will probably need a few weeks more to get things fixed and cleaned up

JoyBed, 2024-04-30:
> There is a PR which is WIP and not up to date : 
> 
> https://github.com/enjoy-digital/litex/pull/1923
> 
> 
> 
> I will probably need a few weeks more to get things fixed and cleaned up

I will try it out. Can you check out the analyser in the files I gave you to see if its okay?

Dolu1990, 2024-04-30:
> I will try it out.

I will let you know when it is ready ^^ don't bother until then.

>  Can you check out the analyser in the files I gave you to see if its okay?

Which file ? i haven't seen any ? Yes i can.

Dolu1990, 2024-04-30:
Ahhhh i did tried them, to check the memory region things.
But can't test anything on hardware.

Dolu1990, 2024-05-02:
@JoyBed Got the vexiiriscv bug fixed, everything seems stable now, remains a bit of timing optimization and then it should be good for usages.

Dolu1990, 2024-05-14:
@JoyBed https://github.com/enjoy-digital/litex/pull/1923 is now good.

Still a few optimisations to do, but take take time ^^

I'm using it to run debian with for instance : 

python3 -m litex_boards.targets.digilent_nexys_video --cpu-type=vexiiriscv  --with-jtag-tap  --bus-standard axi-lite --vexii-args=" \
--allow-bypass-from=0 --debug-privileged --with-mul --with-div --div-ipc --with-rva --with-supervisor --performance-counters 9 \
--regfile-async --xlen=64 --with-rvc --with-rvf --with-rvd --fma-reduced-accuracy \
--fetch-l1 --fetch-l1-ways=4 --fetch-l1-mem-data-width-min=64 \
--lsu-l1 --lsu-l1-ways=4  --lsu-l1-mem-data-width-min=64 --lsu-l1-store-buffer-ops=32 --lsu-l1-refill-count 2 --lsu-l1-writeback-count 2 --lsu-l1-store-buffer-slots=2  --with-lsu-bypass \
--with-btb --with-ras --with-gshare --relaxed-branch"  --cpu-count=4 --with-jtag-tap  --with-video-framebuffer --with-sdcard --with-ethernet --with-coherent-dma --l2-byte=262144 --update-repo=no  --sys-clk-freq 100000000

JoyBed, 2024-05-16:
> @JoyBed #1923 is now good.
>
> Still a few optimisations to do, but take take time ^^
>
> I'm using it to run debian with for instance :
>
> ```
> python3 -m litex_boards.targets.digilent_nexys_video [...]

How many LUTs does a dualcore VexiiRiscv takes ?

Dolu1990, 2024-05-16:
For which ISA ?
RV64IMAFDC to run debian ?
Or something softcore friendly like RV32IMA to just run linux ?

Dolu1990, 2024-05-16:
For RV64IMAFDC single issue but everything enabled with memory coherency, it is around 12K lut, at learly 100 Mhz on Artix 7 -1 (slow speed grade).
The FPU take a lot of space, around 5K lut per core. RVC is also a pain in the ass ^^

JoyBed, 2024-05-16:
> For RV64IMAFDC single issue  everything enabled with memory coherency and single core, it is around 12K lut, at learly 100 Mhz on Artix 7 -1 (slow speed grade).
> 
> The FPU take a lot of space, around 5K lut per core. RVC is also a pain in the ass ^^

Wow, a Debian able core only in 12k LUTs? Thats amazing! I can comfortably fit then even 4 of them in my FPGA!

Dolu1990, 2024-05-16:
Note a recent change in litex broke things XD It works up to litex 86a43c9f

JoyBed, 2024-05-16:
> Note a recent change in litex broke things XD It works up to litex [86a43c9](https://github.com/enjoy-digital/litex/commit/86a43c9ff7141d625a92d75c1e9f5d99bb2d69ab)

Just reverting this will fix things ?

Dolu1990, 2024-05-16:
I mean, you can checkout https://github.com/enjoy-digital/litex/commit/86a43c9ff7141d625a92d75c1e9f5d99bb2d69ab and it will work, but later it will not.

Otherwise there is two commit to revert to get things to work : 
- fbf03ec7
- 26f5e8a1

JoyBed, 2024-05-16:
I dont have these in my local so I dont need to worry about them. My local s like a month old.

JoyBed, 2024-05-16:
> I updated the vexii with a fix. Now it work with upstream litex.

Funny, for some reason I cant checkout your PR. Nevermind tho, I will manually add the needed files.

JoyBed, 2024-05-17:
Also a quick question, does VexiiRiscv keep the offset when talking thru memory bus or not?

Dolu1990, 2024-05-17:
@JoyBed Ahhh right, Vexii was using the old code which was removing the offset.
I just pushed a fix in this PR. should be good now.

JoyBed, 2024-05-17:
> @JoyBed Ahhh right, Vexii was using the old code which was removing the offset.
> 
> I just pushed a fix in this PR. should be good now.

Either way, with VexiiRiscv I have the same problem as with NaxRiscv. Lockup when DRAM is connected thru memory bus but thru peripheral bus it works.

Dolu1990, 2024-05-17:
It lockup only of you try to use more than 512 MB of ram right ?

JoyBed, 2024-05-17:
> It lockup only of you try to use more than 512 MB of ram right ?

No. At any amount of DRAM. I tried from 32 megs all they way up to 1gig.

JoyBed, 2024-05-17:
Now when I was checking it out, why the args for FPU and RVC are commented out? I was wondering why linux are not booting. Also aint it done like in the old VexRiscv where FPU is configurable to be shared between cores?

Dolu1990, 2024-05-17:
> Now when I was checking it out, why the args for FPU and RVC are commented out?

Do you mean things around : 
https://github.com/SpinalHDL/VexiiRiscv/blob/4d2ff4b29d04bf033239ace06fb0f61b3600362d/src/main/scala/vexiiriscv/Param.scala#L163 ? 

It is only for debug, and isn't enabled for litex.
See 

```scala
  //  Debug modifiers
  val debugParam = sys.env.getOrElse("VEXIIRISCV_DEBUG_PARAM", "0").toInt.toBoolean
  if(debugParam) {
```

As long as you feed litex with : 

```
python3 -m litex_boards.targets.digilent_nexys_video --cpu-type=vexiiriscv  --with-jtag-tap  --bus-standard axi-lite --vexii-args=" \
--allow-bypass-from=0 --debug-privileged --with-mul --with-div --div-ipc --with-rva --with-supervisor --performance-counters 9 \
--regfile-async --xlen=64 --with-rvc --with-rvf --with-rvd --fma-reduced-accuracy \
--fetch-l1 --fetch-l1-ways=4 --fetch-l1-mem-data-width-min=64 \
--lsu-l1 --lsu-l1-ways=4  --lsu-l1-mem-data-width-min=64 --lsu-l1-store-buffer-ops=32 --lsu-l1-refill-count 2 --lsu-l1-writeback-count 2 --lsu-l1-store-buffer-slots=2  --with-lsu-bypass \
--with-btb --with-ras --with-gshare --relaxed-branch"  --cpu-count=4 --with-jtag-tap  --with-video-framebuffer --with-sdcard --with-ethernet --with-coherent-dma --l2-byte=262144 --update-repo=no  --sys-clk-freq 100000000  --build
```

It should be debian ready

> Also aint it done like in the old VexRiscv where FPU is configurable to be shared between cores?

Right, it isn't.
Currently, the FPU is thigtly integrated into the pipeline via some plugins : 
https://github.com/SpinalHDL/VexiiRiscv/blob/4d2ff4b29d04bf033239ace06fb0f61b3600362d/src/main/scala/vexiiriscv/Param.scala#L660

It would be great to have an lighter alternative (with less performance).
I was thinking that maybe a FSM like FPU (instead of fully pipelined) would allow reuse a lot of hardware.
Or FPU sharing like in VexRiscv, yes.
I implemented things like they are now as  i wanted to aim at full perf thigtly coupled FPU, which would also work great in ASIC.

Through peripheral memory, did you got linux to work ?

On my side, quad core debian is very very stable, did a lot of test with it. Also, things like USB host / bluetooth / sdcard / ethernet are working well.

JoyBed, 2024-05-17:
> Do you mean things around :
>
> https://github.com/SpinalHDL/VexiiRiscv/blob/4d2ff4b29d04bf033239ace06fb0f61b3600362d/src/main/scala/vexiiriscv/Param.scala#L163 ?
>
>
>
> It is only for debug, and isn't [...]

Yes, thru peripheral bus its working. But the arguments --with-rvc --with-with-rvf --with-rvd gave error as not recognised so I checked the core.py and they are commented out. Also some lines werent in right order in the core.py file so i reorganised it a bit. When I get home I will send you the core.py I modified from your original one in the PR.

Dolu1990, 2024-05-17:
> --with-rvc --with-with-rvf --with-rvd

Not reconized by the python litex itself ? or SpinalHDL generation ?
note that --with-rvc --with-with-rvf --with-rvd are part of the --vexii-args, they aren't directly feed to litex itself
--other-args --vexii-args=" ...  --with-rvc --with-with-rvf --with-rvd ... " --other-args

JoyBed, 2024-05-18:
> Not reconized by the python litex itself ? or SpinalHDL generation ? note that --with-rvc --with-with-rvf --with-rvd are part of the --vexii-args, they aren't directly feed to litex itself [...]

Yes, they are not recognised by the SpinalHDL.
[image]

Dolu1990, 2024-05-18:
Ahhh i think you have a old version of VexiiRiscv then.
I gived you the command with --update-repo=no, i didnt noticed it.
--update-repo=recommended

That may explain a lot.

Dolu1990, 2024-05-18:
Then the pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/ext/vexiiriscv should be on :  "fpu_internal", "8a239d10" (after running the soc generation)

JoyBed, 2024-05-18:
Yes, but with the core.py version in the PR the update_repo doesnt work, I fixed it and also removed the dependancy on NaxRiscv in the process. Yes, now the pythondata are on "fpu_internal". Can we communicate on some other platform so we will get things done faster so this issue can be closed faster ?

[5 more replies]