enjoy-digital/litex #1933: Loading binary hangs up

roby2014, 2024-04-20:
Hi. Trying to simulate my rust binary with:
```
riscv64-elf-objcopy mybin -O binary mybin.bin
litex_sim --output-dir=target/litex_sim --cpu-variant=minimal --rom-init=mybin.bin
```

however, when running it, the litex gets stuck around here:

```
.....

- V e r i l a t i o n   R e p o r t: Verilator 5.024 2024-04-05 rev UNKNOWN.REV
- Verilator: Built from 0.480 MB sources in 5 modules, into 0.776 MB in 22 C++ files needing 0.000 MB
- Verilator: Walltime 0.167 s (elab=0.021, cvt=0.104, bld=0.000); cpu 0.000 s on 1 threads; alloced 15.531 MB
make -j -C /home/roby/repos/colorlight-litex-rs/app/target/litex_sim/gateware/obj_dir -f Vsim.mk Vsim
make[1]: Entering directory '/home/roby/repos/colorlight-litex-rs/app/target/litex_sim/gateware/obj_dir'
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -Os -c -o veril.o /home/roby/litex/litex/litex/build/sim/core/veril.cpp
g++  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -Os -c -o sim_init.o ../sim_init.cpp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -c -o verilated.o /usr/share/verilator/include/verilated.cpp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -c -o verilated_dpi.o /usr/share/verilator/include/verilated_dpi.cpp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -c -o verilated_vcd_c.o /usr/share/verilator/include/verilated_vcd_c.cpp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -c -o verilated_threads.o /usr/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /usr/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vsim.cpp Vsim___024root__DepSet_h104c642d__0.cpp Vsim___024root__DepSet_hb1836b75__0.cpp Vsim_sim__DepSet_h837b84dc__0.cpp Vsim_sim__DepSet_h40728c06__0.cpp Vsim_VexRiscv__DepSet_hda50bfa8__0.cpp Vsim_VexRiscv__DepSet_h9f7c89a9__0.cpp Vsim__Dpi.cpp Vsim__Trace__0.cpp Vsim__ConstPool_0.cpp Vsim___024root__Slow.cpp Vsim___024root__DepSet_h104c642d__0__Slow.cpp Vsim___024root__DepSet_hb1836b75__0__Slow.cpp Vsim_sim__Slow.cpp Vsim_sim__DepSet_h837b84dc__0__Slow.cpp Vsim_sim__DepSet_h40728c06__0__Slow.cpp Vsim_VexRiscv__Slow.cpp Vsim_VexRiscv__DepSet_hda50bfa8__0__Slow.cpp Vsim_VexRiscv__DepSet_h9f7c89a9__0__Slow.cpp Vsim__Syms.cpp Vsim__Trace__0__Slow.cpp Vsim__TraceDecls__0__Slow.cpp > Vsim__ALL.cpp
echo "" > Vsim__ALL.verilator_deplist.tmp
g++ -Os  -I.  -MMD -I/usr/share/verilator/include -I/usr/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-shadow -Wno-sign-compare -Wno-tautological-compare -Wno-uninitialized -Wno-unused-but-set-parameter -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable    -ggdb -Wall -O3   -I/home/roby/litex/litex/litex/build/sim/core   -c -o Vsim__ALL.o Vsim__ALL.cpp
Archive ar -rcs Vsim__ALL.a Vsim__ALL.o
g++    veril.o sim_init.o verilated.o verilated_dpi.o verilated_vcd_c.o verilated_threads.o Vsim__ALL.a   libdylib.o modules.o pads.o parse.o sim.o -lpthread -Wl,--no-as-needed -ljson-c -lz -lm -lstdc++ -Wl,--no-as-needed -ldl -levent  -pthread -lpthread -latomic   -o Vsim
rm Vsim__ALL.verilator_deplist.tmp
make[1]: Leaving directory '/home/roby/repos/colorlight-litex-rs/app/target/litex_sim/gateware/obj_dir'
make: Leaving directory '/home/roby/repos/colorlight-litex-rs/app/target/litex_sim/gateware'

[serial2console] loaded (0x63cfd09a42f0)
[clocker] loaded
[xgmii_ethernet] loaded (0x63cfd09a42f0)
[gmii_ethernet] loaded (0x63cfd09a42f0)
[jtagremote] loaded (0x63cfd09a42f0)
[ethernet] loaded (0x63cfd09a42f0)
[serial2tcp] loaded (0x63cfd09a42f0)
[spdeeprom] loaded (addr = 0x0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0

```

Any idea how I can debug this?

AndrewD, 2024-04-20:
See comment on #1935, but you should be getting a vcd file you can view in gtkwave. Look at the main bus address and data to see what your code is up to.
You could maybe use renode too and get full debug access to your code.

roby2014, 2024-04-20:
I'm able to simulate it with 
> `litex_sim --output-dir=target/litex_sim --cpu-type=vexriscv --rom-init=$1.bin --no-compile-software`

however, not with `--integrated-main-ram-size=0x4000 --ram-init=$1.bin`

How is this causing the issue?

roby2014, 2024-04-20:
Interesting...
```
litex_sim --output-dir=target/litex_sim --cpu-type=vexriscv --rom-init=$1.bin --no-compile-software --gtkwave-savefile

[roby@thonkpad firmware]$ gtkwave target/litex_sim/gateware/sim.vcd

GTKWave Analyzer v3.3.118 (w)1999-2023 BSI

Near byte 49154, VCD search table NULL..is this a VCD file?
No symbols in VCD file..is it malformed?  Exiting!
```