enjoy-digital/litex #1935: Trying to interact with UART via Rust binary

roby2014, 2024-04-20:
Hello. Not sure if the right place to ask, but here I go:

been trying to flash a rust program to my ecp5 fpga (colorlight 5a-75e). however when booting, im not getting the expected output.
```rs
#![no_std]
#![no_main]

extern crate panic_halt;

use riscv_rt::entry;
use litex_pac as pac;

fn uart_write(uart: &pac::Uart, value: u8) {
    while uart_txfull_read(uart) != 0 {}
    uart.rxtx().write(|w| unsafe { w.bits(value.into()) });
    uart.ev_pending().write(|w| unsafe { w.bits(0x1) });
}

fn uart_txfull_read(uart: &pac::Uart) -> u8 {
    return uart.txfull().read().bits() as u8;
}

fn hprint(uart: &pac::Uart, s: &str) {
    for c in s.bytes() {
        uart_write(uart, c);
    }
}

#[entry]
fn main() -> ! {
    let peripherals = unsafe { pac::Peripherals::steal() };
    let uart = peripherals.uart;

    loop {
        hprint(&uart, "a");
    }
}
```

however, it gets stuck at liftoff:
```
litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 20 2024 19:46:07
 BIOS CRC passed (8472a90b)

 LiteX git sha1: c2fd1e9a

--=============== SoC ==================--
CPU:            VexRiscv @ 25MHz
BUS:            wishbone 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32.0KiB
SRAM:           8.0KiB
MAIN-RAM:       16.0KiB

--========== Initialization ============--
Memtest at 0x40000000 (16.0KiB)...
  Write: 0x40000000-0x40004000 16.0KiB   
   Read: 0x40000000-0x40004000 16.0KiB   
Memtest OK
Memspeed at 0x40000000 (Sequential, 16.0KiB)...
  Write speed: 37.7MiB/s
   Read speed: 19.0MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading target/riscv32i-unknown-none-elf/debug/app.bin to 0x40000000 (7876 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload complete (10.0KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

any idea how I can debug this? ive tried uploading a c binary ([this one](https://github.com/roby2014/risc-v-colorlight-5a-75e/blob/master/firmware/main.c) and works fine.

this is how im building my soc:

```py
class BaseSoC(SoCCore):
    def __init__(self, version, revision):
        # 25 MHz
        sys_clk_freq = int(25e6)

        # SoC with CPU
        platform = colorlight_5a_75e.Platform(revision)
        SoCCore.__init__(self, platform,
            cpu_type                 = "vexriscv",
            clk_freq                 = sys_clk_freq,
            ident                    = f"LiteX RISC-V CPU Test SoC {version}", ident_version=True,
            integrated_rom_size      = 0x8000,
            integrated_main_ram_size = 0x4000)

        # Clock Reset Generation
        self.submodules.crg = CRG(platform.request("clk25"))

# Build --------------------------------------------------------------------------------------------

def main():
    parser = argparse.ArgumentParser(description="LiteX RISC-V SoC on Colorlight 5A-75E")
    builder_args(parser)
    soc_core_args(parser)
    trellis_args(parser)
    parser.add_argument("--build", action="store_true", help="Build bitstream")
    parser.add_argument("--load",  action="store_true", help="Load bitstream")
    parser.add_argument("--cable", default="ft232",    help="openFPGALoader JTAG probe model")
    parser.add_argument("--revision", default="6.0",  help="Colorlight 5A-75E model revision")
    args = parser.parse_args()

    soc = BaseSoC("5A-75E", revision=args.revision)

    builder = Builder(soc, **builder_argdict(args))
    builder.build(**trellis_argdict(args), run=args.build)

    if args.load:
        extra_args = ""
        if args.cable == "ft232RL":
            extra_args = "--pins=RXD:RTS:TXD:CTS"
        elif args.cable == "usb-blaster":
            extra_args = f"--probe-firmware {os.environ['QUARTUSPATH']}"
            
        bitstream_file = os.path.join(builder.gateware_dir, f'{soc.build_name}.bit')
        cmd = f"openFPGALoader --cable {args.cable} {extra_args} {bitstream_file}"
        print(f"Uploading bitstream file: {bitstream_file}")
        print(f"JTAG cable: {args.cable}")
        print(f"Running command: {cmd}")
        os.system(cmd)

if __name__ == "__main__":
    main()

```

AndrewD, 2024-04-20:
You could use litescope to capture the soc bus address to see where your code is going. But if you use the sim as per your other issue you are probably already getting a vcd file that shows the full internal state of the design.

roby2014, 2024-04-21:
Program works in simulation with

> `litex_sim --output-dir=target/litex_sim --cpu-type=vexriscv --rom-init=$1.bin --no-compile-software --csr-csv "csr.csv"`

Hmm...

enjoy-digital, 2024-06-13:
Hi @roby2014,

do you still have the issue? If working in simulation from ROM, you could also try to run it from MAIN_RAM (`--ram-init`) in simulation, this way you'll be in condition very similar to your hardware test). I don't have that much experience with Rust, but is your program configured to run from MAIN_RAM (`0x4000_0000`). If working in simulation in these condition, I expect it to also work on hardware. If not working, you'll have the full visibility on the SoC signals with `--trace`

roby2014, 2024-06-13:
hi @enjoy-digital . I've not looked at this since I created this issue.

so, running the above program in simulation work with `rom` init, but does not with `ram`, so im supposing the hardware issue is the same as the simulation + ram.

Please see the following logs if they help:

## --rom-init

```
set -e
riscv64-elf-objcopy $1 -O binary $1.bin
litex_sim --output-dir=target/litex_sim --cpu-type=vexriscv --rom-init=$1.bin --no-compile-software --csr-csv "csr.csv"
```

output:
```
(.venv) [roby@thonkpad firmware]$ cargo run
   Compiling litex-pac v0.1.0 (/home/roby/repos/colorlight-litex-rs/litex-pac)
   Compiling firmware v0.1.0 (/home/roby/repos/colorlight-litex-rs/firmware)
    Finished dev [unoptimized + debuginfo] target(s) in 0.52s
     Running `/home/roby/repos/colorlight-litex-rs/firmware/.cargo/sim.sh target/riscv32i-unknown-none-elf/debug/firmware`
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-06-13 23:27:52)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 1.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv added.
INFO:SoC:CPU vexriscv adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode:  RW, Cached: False, Linker: False.
INFO:SoC:CPU vexriscv overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoC:CPU vexriscv adding Interrupt(s).
INFO:SoC:CPU vexriscv adding SoC components.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoC:        __   _ __      _  __  
[... 693 lines cut ...]
[serial2tcp] loaded (0x5911855162f0)
[spdeeprom] loaded (addr = 0x0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0
```

In rust, the HAL & PAC auto generate  `memory.x` which maps the regions:
```
MEMORY {
	rom : ORIGIN = 0x00000000, LENGTH = 0x00008000
	sram : ORIGIN = 0x10000000, LENGTH = 0x00002000
	main_ram : ORIGIN = 0x40000000, LENGTH = 0x00004000
	bootrom : ORIGIN = 0x20000000, LENGTH = 0x00000400
	csr : ORIGIN = 0xf0000000, LENGTH = 0x00010000
}

REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_RODATA", rom);
REGION_ALIAS("REGION_DATA", sram);
REGION_ALIAS("REGION_BSS", sram);
REGION_ALIAS("REGION_HEAP", sram);
REGION_ALIAS("REGION_STACK", sram);

/* CPU reset location. */
_stext = 0x000000;
```

AndrewD, 2024-07-08:
`MEMORY {
	rom : ORIGIN = 0x00000000, LENGTH = 0x00008000
	sram : ORIGIN = 0x10000000, LENGTH = 0x00002000
	main_ram : ORIGIN = 0x40000000, LENGTH = 0x00004000
	bootrom : ORIGIN = 0x20000000, LENGTH = 0x00000400
	csr : ORIGIN = 0xf0000000, LENGTH = 0x00010000
}

REGION_ALIAS("REGION_TEXT", rom);
REGION_ALIAS("REGION_RODATA", rom);
REGION_ALIAS("REGION_DATA", sram);`

The problem is that you code is linked into rom. Edit memory.x to place code in main_ram as follows:
`
REGION_ALIAS("REGION_TEXT", main_ram);
REGION_ALIAS("REGION_RODATA", main_ram);
REGION_ALIAS("REGION_DATA", sram);`

Then build your binary to load to main ram.

I don't think the reset address is used when you load this way.

It would be great if you can document the steps you followed to get to here as you are very close to a working solution: from a quick search I suspect you used svd2rust and possibly followed an online write-up I found, but it would be great to get confirmation. 

I'm interested in reproducing this and adding some info to the wiki.

roby2014, 2024-09-26:
1. i load soc with `python3 soc.py --load --no-compile-software --memory-x "../litex-pac/memory.x"`
2. i change `memory.x` (`main_ram` for text, `rodata` and `data` and remove `_stext = 0x000000;`)
3. i flash firmware with `litex_term --kernel firmware.bin /dev/ttyUSB0`
4. nothing happens
5. i load soc again with same command from step 1, i see expected result (hello world)
6. everytime i load firmware with command from step 3 , everything works fine

is this expected? shouldnt it work at the first time i flash the firmware?