enjoy-digital/litex #1947: RISC-V debug module support for CV32E40P/41P ?

mohamedElbouazzati, 2024-05-03:
Hello @enjoy-digital, 

I wanted to utilize the risc-v debug module provided in `core.py` within `litex/litex/soc/cores/cpu/cv32e40p` https://github.com/enjoy-digital/litex/blob/76a704377fbd2897a5ab7144419fe0733aef3d97/litex/soc/cores/cpu/cv32e40p/core.py#L283C1-L283C32. 
However, I encountered an issue when running OpenOCD:
`Error: [riscv.cpu] Unable to halt. dmcontrol=0x80000001, dmstatus=0x00000c82`
Could you please confirm whether I am missing something?
Below are further details regarding the integration of the debug module in core.py, the target file, and the OpenOCD output.

1. **core.py**

```
class CV32E41P(CPU):
    category             = "softcore"
    family               = "riscv"
    name                 = "cv32e41p"
    human_name           = "CV32E41P"
......
.....
        # added these lines to call the debug module class. 
        self.dm=DebugModule()
        self.add_debug_module(self.dm)
        self.dm.add_sources(platform)  

    def add_debug_module(self, dm):
        self.cpu_params.update(i_debug_req_i=dm.debug_req)
        self.cpu_params.update(i_rst_ni=~(ResetSignal("sys") | dm.ndmreset))  
```

2. **SoC target.py** 

```
        # add jtag
        pin_tdi = platform.request("jtag_tdi")
        pin_tms = platform.request("jtag_tms")
        pin_tck = platform.request("jtag_tck")
        pin_tdo = platform.request("jtag_tdo")

        self.comb += [
        self.cpu.dm.dm_params["i_tdi"].eq(pin_tdi),
        self.cpu.dm.dm_params["i_tms"].eq(pin_tms),
        self.cpu.dm.dm_params["i_tck"].eq(pin_tck),
        self.cpu.dm.dm_params["i_trst_n"].eq(1),
        pin_tdo.eq(self.cpu.dm.dm_params["o_tdo"])
    ]
```
3. **debug module RTL**
[image]

4. Error from openocd
```
Open On-Chip Debugger 0.12.0+dev-03744-g3991492cc (2024-04-24-12:59)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi device_desc' not 'ftdi_device_desc'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 500 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x11001cdf (mfg: 0x66f (lowRISC), part: 0x1001, ver: 0x1)
Info : [riscv.cpu] datacount=2 progbufsize=8
Error: [riscv.cpu] Unable to halt. dmcontrol=0x80000001, dmstatus=0x00000c82
Error: [riscv.cpu] Fatal: Hart 0 failed to halt during examine
Error: [riscv.cpu] Examination failed
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: [riscv.cpu] Unsupported DTM version: -1
Error: [riscv.cpu] Could not identify target type
```
Do you have any ideas on how to debug this issue? 
Many thanks for your assistance :+1:

coolderin, 2024-06-18:
Hi there,
I have same issue. However openocd doesn't detect debug module's TAP. Could you share your openocd configurations?
Thx

mohamedElbouazzati, 2024-06-21:
Hi @coolderin, 

I'm using this one [sw/app/openocd_digilent_hs2.cfg](https://github.com/ThalesGroup/cva6-softcore-contest/blob/cv32a6_contest_23_24/sw/app/openocd_digilent_hs2.cfg)  from cva6-softcore-contest repo.

coolderin, 2024-06-22:
Hi, 
Thanks for the configuration.My mistake was I tried the implement Debug Module outside from CPU. You implement the Debug Module inside the CPU. 
Now I have several issues
First modified openocd configuration.
`interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 44853
reset_config none

transport select jtag

reset_config none

# TAP declaration
set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3

#set _TARGETNAME $_CHIPNAME.cpu
#target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0

gdb_report_data_abort enable
gdb_report_register_access_error enable

#riscv set_reset_timeout_sec 120
#riscv set_command_timeout_sec 120

# prefer to use sba for system bus access
#riscv set_prefer_sba on

log_output openocd_log_without_Target.txt

# dump jtag chain
scan_chain
bindto 0.0.0.0

init
#halt
echo "Ready for Remote Connections"`

1) Prefering system bus access causes error in openocd. 
`openocd -f openocd.cfg 
Open On-Chip Debugger 0.12.0+dev-03811-g568baf8c0 (2024-06-08-16:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'remote_bitbang host' not 'remote_bitbang_host'
DEPRECATED! use 'remote_bitbang port' not 'remote_bitbang_port'
riscv
  riscv authdata_read [index]
  riscv authdata_write [index] value
  riscv dm_read reg_address
  riscv dm_write reg_address value
  riscv dmi_read address
  riscv dmi_write address value
  riscv dump_sample_buf [base64]
  riscv etrigger set [vs] [vu] [m] [s] [u] <exception_codes>|clear
  riscv exec_progbuf instr1 [instr2 [... instr16]]
  riscv expose_csrs n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv expose_custom n0[-m0|=name0][,n1[-m1|=name1]]...
  riscv hide_csrs {n0|n-m0}[,n1|n-m1]......
  riscv icount set [vs] [vu] [m] [s] [u] [pending] <count>|clear
  riscv info
  riscv itrigger set [vs] [vu] [nmi] [m] [s] [u] <mie_bits>|clear
  riscv memory_sample bucket address|clear [size=4]
  riscv repeat_read count address [size=4]
  riscv reset_delays [wait]
  riscv resume_order normal|reversed
  riscv set_bscan_tunnel_ir value
  riscv set_command_timeout_sec [sec]
[... 77 lines cut ...]
    riscv.cpu riscv set_ir [idcode|dtmcs|dmi] value
    riscv.cpu riscv set_maskisr ['off'|'steponly']
    riscv.cpu riscv set_mem_access method1 [method2] [method3]
    riscv.cpu riscv set_reset_timeout_sec [sec]
    riscv.cpu riscv use_bscan_tunnel value [type]
  riscv.cpu set_reg dict
  riscv.cpu smp [on|off]
  riscv.cpu smp_gdb
  riscv.cpu was_examined
  riscv.cpu write_memory address width data ['phys']
openocd.cfg:25: Error: 
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "openocd.cfg", line 25
Error: [riscv.cpu] Unsupported DTM version: -1
Error: [riscv.cpu] Could not identify target type.
`
2) When I disable prefer_sba_access (make this line comment), i got error like yours.
[openocd_log.txt](https://github.com/user-attachments/files/15936356/openocd_log.txt)

3) If i disable target creation and other riscv commands i can't communicate with TAP.
[openocd_log_without_Target.txt](https://github.com/user-attachments/files/15936370/openocd_log_without_Target.txt)

I think maybe cpu/ debug module doesn't have any hardware thread (?).
Does this CPU have to need a trigger module?