enjoy-digital/litex #1950: Issues with main memory > 1GB

hightowera, 2024-05-09:
I'm not sure if this is an actual issue or something I'm missing in the configuration (not sure where to change it).  I'm trying to add a patch (which I would like to upstream) to add support for the TI DCA1000EVM millimeter wave capture board.  It's an ECP5-85K board with two 8-bit 8GBit DDR3L memory parts.  It also has GBit Ethernet, two FT2234s, large SPI flash, LVDS expansion header, and plenty of switches and LEDs.  The memory size is the problem.  The parts are in the MT41K family so it was pretty easy to add a profile in the memory modules.  When I try to configure the SoC, I get the following output:

```
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x80000000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:CSR Bridge csr added.
INFO:SoCBusHandler:csr Region added at Origin: 0xf0000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:csr added as CSR Master.
ERROR:SoCRegion:Origin needs to be aligned on size:
ERROR:SoCRegion:Origin: 0x40000000, Size: 0x80000000, Mode: RWX, Cached:  True, Linker: False
```

It's configuring the 2GBytes of DDR3L at a base address of `0x40000000` (1GB aligned) and then complains about alignment.  I've tried adding `--bus-address-width 64` to the script args, but it doesn't seem to help.  This is wishbone btw.  It seems that even with 32-bit busses, the CSR regs could be moved lower and the 2GB RAM could reside at `0x80000000 -> 0xFFFFFFFF`

I'm happy to look into this further, but thought you might have some initial thoughts first.  Thanks!

enjoy-digital, 2024-05-10:
Hi @hightowera,

the following repository might be useful for you: https://github.com/enjoy-digital/litex_64bit_addressing_test/

enjoy-digital, 2026-10-21:
Closing since the behavior is expected (regions must be aligned to their size) and there was no further feedback. Feel free to reopen if needed.

closed 2026-10-21 by enjoy-digital