enjoy-digital/litex #1960: RFC: refactor devicetree (DTS) generation (draft)

AndrewD, 2024-05-21:
### Current status
The current state of `litex_json2dts_linux.py` is just enough to generate a DTS for {linux_on_litex_vexriscv,rocket} with some of the standard devices, however it is hardcoded to just the standard litex peripherals with linux drivers and does not cater for other litex cores in separate git modules or custom peripherals. Some devices are only partially supported with mostly hard coded configuration, eg custom instances of gpio for [leds](https://github.com/enjoy-digital/litex/blob/10083f4d87aae73599a7d929bd57cda70f8b0371/litex/tools/litex_json2dts_linux.py#L543) and pwm for [rgb_led](https://github.com/enjoy-digital/litex/blob/10083f4d87aae73599a7d929bd57cda70f8b0371/litex/tools/litex_json2dts_linux.py#L551).

### Motivation
To generalise the dts to cater for standard peripherals in a generic way and also support additional peripherals.

### Proposal
TODO

m-byte, 2024-05-23:
It's pretty much the same for the zephyr version of this script. I believe it would be preferable, if these two scripts could be combined as it should be possible to have only minor differences between Zephyr and Linux dts files for LiteX.

As for custom peripherals from separate git repos, being able to support them with the same script would be even better. As it stands, we currently have our own hard coded copy of the zephyr script that we are working with. That's probably what most do, but it makes collaboration harder and (unnecessarily) raises the entry barrier.

m-byte, 2024-05-23:
@maass-hamburg

AndrewD, 2024-05-27:
Description edited, more to do

AndrewD, 2024-06-04:
@enjoy-digital @trabucayre this issue describes the motivation behind the changes in #1975. This is a minimal change to start dynamically generating dts content for litex cores/modules instead of hard coding in the json2dts scripts.
With these changes we have been able to remove all of our local customisations from `litex_json2dts_linux.py`.

In the future a pure python dts library like https://github.com/molejar/pyFDT may be interesting to consider as a more fully featured approach if DTS generation is moved into the builder.