enjoy-digital/litex #1970: OpenOCD error when "make" after "./configure --enable-ftdi"

angerpro1411, 2024-05-29:
Hi, i got probleme when try to install OpenOCD. The error like this after I tried to "make".

```
dell3561-49@dell3561-49:~/Documents/LiteXver2/OpenOCD/openocd$ make
Makefile:5467: warning: overriding recipe for target 'check-recursive'
Makefile:4273: warning: ignoring old recipe for target 'check-recursive'
cat src/helper/startup.tcl src/jtag/startup.tcl src/target/startup.tcl src/server/startup.tcl src/flash/startup.tcl | ./src/helper/bin2char.sh > src/startup_tcl.inc || { rm -f src/startup_tcl.inc; false; }
cp src/jtag/drivers/minidriver_imp.h src/jtag/minidriver_imp.h
make  all-recursive
make[1]: Entering directory '/home/dell3561-49/Documents/LiteXver2/OpenOCD/openocd'
Makefile:5467: warning: overriding recipe for target 'check-recursive'
Makefile:4273: warning: ignoring old recipe for target 'check-recursive'
Making all in jimtcl
make[2]: Entering directory '/home/dell3561-49/Documents/LiteXver2/OpenOCD/openocd/jimtcl'
/usr/bin/tclsh ./make-load-static-exts.tcl aio array clock eventloop exec file history interp load namespace pack package posix readdir regexp signal syslog tclprefix binary glob nshelper oo stdlib tclcompat tree >_load-static-exts.c || ( rm _load-static-exts.c; exit 1)
jim-interp.c: In function ‘JimInterpCommand’:
jim-interp.c:164:46: warning: ‘%ld’ directive output may be truncated writing between 1 and 20 bytes into a region of size 19 [-Wformat-truncation=]
  164 |     snprintf(buf, sizeof(buf), "interp.handle%ld", Jim_GetId(interp));
      |                                              ^~~
jim-interp.c:164:32: note: directive argument in the range [-9223372036854775807, 9223372036854775807]
  164 |     snprintf(buf, sizeof(buf), "interp.handle%ld", Jim_GetId(interp));
      |                                ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:894,
                 from jim.h:74,
                 from jim-interp.c:3:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:71:10: note: ‘__builtin___snprintf_chk’ output between 15 and 34 bytes into a destination of size 32
   71 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   72 |                                    __glibc_objsize (__s), __fmt,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   73 |                                    __va_arg_pack ());
      |                                    ~~~~~~~~~~~~~~~~~
src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
src/flash/nor/jtagspi.c:249:29: error: ‘status’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  249 |                 if ((status & SPIFLASH_BSY_BIT) == 0) {
      |                     ~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:22,
                 from src/flash/nor/jtagspi.c:22:
src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
./src/helper/log.h:130:9: error: ‘status’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  130 |         log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |         ^~~~~~~~~~~~~
src/flash/nor/jtagspi.c:262:18: note: ‘status’ was declared here
  262 |         uint32_t status;
      |                  ^~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3341: src/flash/nor/jtagspi.lo] Error 1
make[2]: Leaving directory '/home/dell3561-49/Documents/LiteXver2/OpenOCD/openocd'
make[1]: *** [Makefile:4273: all-recursive] Error 1
make[1]: Leaving directory '/home/dell3561-49/Documents/LiteXver2/OpenOCD/openocd'
make: *** [Makefile:1997: all] Error 2
```
Somebody got the same error? I really don't know how to fix it

Dolu1990, 2024-05-29:
Hi,

To me it seems like this kind of thing often happen with openocd, issues emerge when using new recent GCC versions against "old" openocd sources.
what gcc version do you have ?

angerpro1411, 2024-05-29:
my gcc --version is 11.4.0 under Ubuntu

Dolu1990, 2024-05-31:
hmm seems not to recent.

I don't know then.
What openocd git / hash did you got ?