enjoy-digital/litex #1986: fatal error: json-c/json.h: No such file or directory

wojtess, 2024-06-12:
While using command `litex_sim --cpu-type=vexriscv` I have folling error:
```
/home/wojtess/programs/litex/litex/litex/build/sim/core/modules/xgmii_ethernet/xgmii_ethernet.c:11:10: fatal error: json-c/json.h: No such file or directory
   11 | #include <json-c/json.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```
I tired these commands:
`sudo apt-get install libjsoncpp-dev` and `sudo ln -s /usr/include/jsoncpp/json/ /usr/include/json`

OS: `Linux wojtess-B450-AORUS-ELITE 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
whole error:
```
[...]
 CC       cmd_litedram.o
 CC       cmd_liteeth.o
 CC       cmd_litesdcard.o
 CC       cmd_litesata.o
 CC       sim_debug.o
 CC       main.o
 CC       crt0.o
 CC       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.crcfbigen bios.bin --little
python3 -m litex.soc.software.memusage bios.elf /home/wojtess/programs/litex/build/sim/software/bios/../include/generated/regions.ld riscv64-unknown-elf

ROM usage: 23.11KiB     (18.05%)
RAM usage: 1.62KiB      (20.21%)

rm crt0.o
make: Leaving directory '/home/wojtess/programs/litex/build/sim/software/bios'
INFO:SoC:Initializing ROM rom with contents (Size: 0x5c80).
INFO:SoC:Auto-Resizing ROM rom from 0x20000 to 0x5c80.
Traceback (most recent call last):
  File "/home/wojtess/.local/bin/litex_sim", line 33, in <module>
    sys.exit(load_entry_point('litex', 'console_scripts', 'litex_sim')())
  File "/home/wojtess/programs/litex/litex/litex/tools/litex_sim.py", line 560, in main
    builder.build(
  File "/home/wojtess/programs/litex/litex/litex/soc/integration/builder.py", line 414, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/wojtess/programs/litex/litex/litex/soc/integration/soc.py", line 1463, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/wojtess/programs/litex/litex/litex/build/sim/platform.py", line 57, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/wojtess/programs/litex/litex/litex/build/sim/verilator.py", line 272, in build
    _compile_sim(build_name, verbose)
  File "/home/wojtess/programs/litex/litex/litex/build/sim/verilator.py", line 169, in _compile_sim
    raise OSError("Subprocess failed with {}\n{}".format(p.returncode, "\n".join(error_messages)))
OSError: Subprocess failed with 2
make: Entering directory '/home/wojtess/programs/litex/build/sim/gateware'
mkdir -p modules
make -C modules -f /home/wojtess/programs/litex/litex/litex/build/sim/core/modules/Makefile
make[1]: Entering directory '/home/wojtess/programs/litex/build/sim/gateware/modules'
mkdir -p xgmii_ethernet
make MOD=xgmii_ethernet -C xgmii_ethernet -f /home/wojtess/programs/litex/litex/litex/build/sim/core/modules/xgmii_ethernet/Makefile
make[2]: Entering directory '/home/wojtess/programs/litex/build/sim/gateware/modules/xgmii_ethernet'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/wojtess/programs/litex/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -I/home/wojtess/programs/litex/litex/litex/build/sim/core/modules/xgmii_ethernet/../.. -o xgmii_ethernet.o /home/wojtess/programs/litex/litex/litex/build/sim/core/modules/xgmii_ethernet/xgmii_ethernet.c
/home/wojtess/programs/litex/litex/litex/build/sim/core/modules/xgmii_ethernet/xgmii_ethernet.c:11:10: fatal error: json-c/json.h: No such file or directory
   11 | #include <json-c/json.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/home/wojtess/programs/litex/litex/litex/build/sim/core/modules/rules.mak:26: xgmii_ethernet.o] Error 1
make[2]: Leaving directory '/home/wojtess/programs/litex/build/sim/gateware/modules/xgmii_ethernet'
make[1]: *** [/home/wojtess/programs/litex/litex/litex/build/sim/core/modules/Makefile:9: xgmii_ethernet] Error 2
make[1]: Leaving directory '/home/wojtess/programs/litex/build/sim/gateware/modules'
make: *** [/home/wojtess/programs/litex/litex/litex/build/sim/core/Makefile:69: modules] Error 2
make: Leaving directory '/home/wojtess/programs/litex/build/sim/gateware'```

enjoy-digital, 2024-06-13:
@wojtess: `libjson-c-dev` install seems to be missing.

wojtess, 2024-06-13:
You are right, that fixed problem, thank you!

closed 2024-06-13 by wojtess