enjoy-digital/litex #2014: Error building design without CPU for UP5K

jamesdiacono, 2024-07-11:
On my machine, the command `./lattice_ice40up5k_evn.py --cpu-type None --build` gives

```
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-07-10 21:41:07)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : ice40-up5k-sg48.
INFO:SoC:System clock: 12.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU None added.
INFO:SoC:CPU None adding IO Region 0 at 0x00000000 (Size: 0x100000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x00000000, Size: 0x100000000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:Allocating Cached Region of size 0x00020000...
INFO:SoCRegion:Region size rounded internally from 0xffffffff to 0x100000000.
INFO:SoCBusHandler:sram Region allocated at Origin: 0x00000000, Size: 0x00020000, Mode:  RW, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoCBusHandler:spiflash Region added at Origin: 0x10000000, Size: 0x00400000, Mode:  RW, Cached:  True, Linker: False.
INFO:SoCBusHandler:spiflash added as Bus Slave.
INFO:SoCBusHandler:rom Region added at Origin: 0x10020000, Size: 0x00008000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoC:CSR Bridge csr added.
ERROR:SoCBusHandler:Region overlap between sram and csr:
ERROR:SoCBusHandler:Origin: 0x00000000, Size: 0x00020000, Mode:  RW, Cached:  True, Linker: False
ERROR:SoCBusHandler:Origin: 0x00000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False
```

and an exit code of 1. On the other hand, I was able to run `./logicbone.py --cpu-type None --build` with no problems.

I am using macOS 14.5 and Python 3.9.6.

enjoy-digital, 2024-07-17:
Thanks @jamesdiacono for the feedback, this should be fixed with https://github.com/litex-hub/litex-boards/commit/88ab3eca6fa07a8464fc0616a62fa8baa9232c94.

jamesdiacono, 2024-07-17:
Solved! Thanks very much.

closed 2024-07-17 by jamesdiacono