enjoy-digital/litex #2020: Issues with compiling and loading a program

phillippZZ, 2024-07-23:
hi, everyone. I'm trying to run a program to test the speed of reading/writing of L1, L2 cache and memory of naxriscv.

What I tried is to put the test program into the rootfs.cpio file downloaded from "https://github.com/litex-hub/linux-on-litex-vexriscv/files/8331338/linux_2022_03_23.zip", and then load all the files to the board with litex_term.

It didn't work well because when I use the following instructions to install riscv-gnu-toolchain, it fails.
>./configure --prefix=/opt/riscv --enable-multilib
>make linux

I also tried a docker container with old version of riscv-gnu-toolchain, which could compile my program. But I got the following error when running the program on the linux system from  "https://github.com/litex-hub/linux-on-litex-vexriscv/files/8331338/linux_2022_03_23.zip".
>root@buildroot:~# ./test
./test: /lib32/libc.so.6: version `GLIBC_2.29' not found (required by ./test)

I think this is because the /lib32/libc.so.6 on the linux system has a incompatible version:
>GNU C Library (Buildroot) stable release version 2.34.
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 10.3.0.
libc ABIs: UNIQUE ABSOLUTE IFUNC
...

So, is there any solution to compiling my program in the correct way?
Also, it would be okay to run my program in bare-metal  environment. But I don't know how to use libraries like <stdio.h> and <time.h> in this scenario.

AndrewD, 2024-07-23:
The [demo](https://github.com/enjoy-digital/litex/blob/c51d22074f6815679942ed9968b7194be33d4f8a/litex/soc/software/demo/README.md#L4) shows how to build bare metal code as a start. Does this work for you?

phillippZZ, 2024-07-23:
I believe bare metal code could work. But I can't find the source code of it.
May I ask how I can modify and compile the source code? I guess loading the .bin file with litex_term won't be a problem.

closed 2024-07-26 by phillippZZ