enjoy-digital/litex #2023: can't use "litex_bare_metal_demo"

phillippZZ, 2024-07-27:
Hey, guys. 

I'm following "https://github.com/enjoy-digital/litex/tree/master/litex/soc/software/demo" and trying to use "litex_bare_metal_demo". But I failed after finishing the installation of litex.

Then I ran the "demo.py" directly and got the following error:

[image]

1.Could anyone teach me how to generat demo.bin correctly? My target is to test cache and memory read/write speed. So I need to use functions like printf and time.

2.I also tried to compile my test program with different riscv-linux-gnu tools and put it into the .cpio file from "https://github.com/litex-hub/linux-on-litex-vexriscv/files/8331338/linux_2022_03_23.zip". But I can't invoke the program on the linux system no matter how.
[image]

AndrewD, 2024-07-27:
`--build-path` is the path to the litex soc you have built for your target. It looks like you aren't pointing `--build-dir` to the correct location. The readme you linked has two steps that must be done in order:

1) Build the soc. in this example it will generate output in `./build/digilent_arty`
`python3 -m litex_boards.targets.digilent_arty --build --load`
2) Build the demo against the soc. 
`litex_bare_metal_demo --build-path=build/digilent_arty`
This expects to find this file generated by the build, in this case: `build/digilent_arty/software/include/generated/variables.mak`

You can change the target, but `--build-dir` must match the target you have already successfully built.

phillippZZ, 2024-07-27:
Now the board stucks at lift-off. 
[image]
I'm using the bitstream successfully built for xilinx_vcu118 before. But now I'm not using vivado, so "python3 -m litex_boards.targets.xilinx_vcu118 --build --load --cpu-type=naxriscv" doesn't finish and fails after first several steps. Even so, the build of demo.bin seems fine since variables.mak could be generated.
But I'm not sure if the incompleteness of bitstream building stage causes the failure of booting up.

closed 2024-07-30 by phillippZZ