enjoy-digital/litex #2027: How to utilize multiple cores

phillippZZ, 2024-07-31:
Hi, everyone.

I successfully deploy bitstream for multiple naxriscv cores onto my xilinx_vcu118 board and am running demo.bin on it. The issue is that I don't know how to let all the cores call my read_speed_test()/write_speed_test() function. Based on the similar results I get, only 1 core is being used no matter how many cores are at present.

Dolu1990, 2024-07-31:
Hi,
the way things works is : 
- In the litex bios, only one core is running.
- Once it exit the litex bios, all CPUs will run the user application, starting from its entry point.

which mean you application need to handle all those cores in assembly before letting them go to C code. 

here is an example of SMP demo, not related to litex : 
https://github.com/SpinalHDL/SaxonSoc/tree/dev-0.3/software/standalone/smpDemo

phillippZZ, 2024-08-03:
@Dolu1990 Thank you, Dolu.

Could you tell me where to find the addresses for all the harts? It seems that smp_unlock(smpInit) is not provided in the generated files under build/ directory after I build bitstream for the board.

Dolu1990, 2024-08-03:
If you have 4 cores, they will be numbered from 0 to 3 in the mhartid CSR

Be sure you have the SMP symbole defined : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/common/start.S#L14

phillippZZ, 2024-08-04:
@Dolu1990 hi,
Now I'm using the following code:
>smpInit:
    // Initialize the stack of the given hart
    csrr a0, mhartid         // Read hart id into a0
    li a1, STACK_PER_HART    // Load stack size into a1
    addi a0, a0, 1           // Increment hart id by 1
    mul a0, a0, a1           // Multiply by stack size to get stack address offset
    la a2, 0x45000000        // Load base address of stack into a2
    add a2, a2, a0           // Add offset to base address
    addi sp, a2, 0          // Set stack pointer to hart-specific stack
    call halt
    // Call the common function
    call commonFunction     // Call commonFunction

If halt() is called for all the secondary harts, the program can execute successfully. If commonFunction() is called, the program breaks and even hart0 does nothing.
The code for hart0 is as following:
>int main(void) {
#ifdef CONFIG_CPU_HAS_INTERRUPT
    irq_setmask(0);
    irq_setie(1);
#endif
    uart_init();
    printf(
        "%s@%ldMHz\n",
        CONFIG_CPU_HUMAN_NAME, CONFIG_CLOCK_FREQUENCY / 1000000
    );    
    // smpInit();
    smp_unlock((uint32_t)&smpInit); // Bring other harts to life
    commonFunction(); // Run common function for hart 0
    return 0;
}

Dolu1990, 2024-08-05:
> Are all the harts sharing the same stack pointer insteading of their own?

The baremetal software crt.S need to give one dedicated stack per core
"addi sp, a2, 0 // Set stack pointer to hart-specific stack"

Did you gived a different hart ID to each nax core ?

Dolu1990, 2024-08-05:
Those hartid are set durring the hardware generation, and are readonly by the software.

> My code sets stack pointers based on the assumption that all the harts have their own IDs already

Yes right, I was confused by thinking it was a pure NaxRiscv github issue, but as you use the NaxRiscv in litex, this one is already set. So all good on that side.

What is your stack_size value ?

phillippZZ, 2024-08-05:
Hope it helps. I tried to set hartStackBase and STACK_PER_HART with different values(big and small), they don't work.
>hartStackBase:
    .word 0x60000000  // Base address for hart stacks
STACK_PER_HART:    
    .word 4096  // Stack size for each hart

Dolu1990, 2024-08-05:
I don't have time to debug your code, but overall, you need to : 
- give every hard a different stack pointer
- put the cores with non zero hardid in a loop lock. while the core with hartid zero initialize things.
- make the hartid core 0 unlock the loop lock.

Dolu1990, 2024-08-05:
Ahh there was also this file : 
https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/smpDemo/src/smpInit.S
But it seems like you already found it XD

phillippZZ, 2024-08-05:
Thank you, @Dolu1990 

I just tried to run linux image on the 4core and 8core bitstream and when I use nproc command, I got "2" for both of them.
[image]

I used "python3 -m litex_boards.targets.xilinx_vcu118 --build --cpu-type=naxriscv  --l2-size=0 --cpu-count=4/8 --timer-uptime" to generate the bitstream. I'm thinking if "--timer-uptime" is bringing some trouble.

Dolu1990, 2024-08-05:
Did you updated the device tree ? (dts -> dtb)
Was the opensbi compiled to support more than 2 harts ?

phillippZZ, 2024-08-05:
Oh, my bad. I forgot to update dts....

Dolu1990, 2024-08-05:
One thing, it is fine to have a dts with more cores than you realy have

phillippZZ, 2024-08-05:
Got it! it would be convenient. I was modifying dts all the time.

But I still can't run the bare metel program. I'm using the exact files from "https://github.com/SpinalHDL/SaxonSoc/blob/dev-0.3/software/standalone/smpDemo/src" with a little bit modification, but it still doesn't work. As long as more than 1 hart is working(changing stack pointer), it will crash. 

I'll test the linux image first. If the linux is working fine, I'll have more attempts on the bare metal.

phillippZZ, 2024-08-05:
Now I'm using the following initialization function for all the secondary harts:
>smpInit:
    # Load the address of hartStack into a0
    li sp, hartStack

    csrr a0, mhartid  # Read hart ID into a0
    li a1, STACK_PER_HART  # Load the value of STACK_PER_HART (1024) into a1
    mul a0, a0, a1         # Multiply hart ID by 1024
    add sp, sp, a0         # Add the result to sp

    // Call common function
    call commonFunction     
    call halt
    ret                     // Return from smpInit

>void commonFunction() {
    printf("test here\n");
}
If there's only one hart running, the printf() in commonFunction() works. But as long as there're more than 1 hart running, the program crashes. I literally have no idea what's going on... Also, I have to let hart0(the main hart) halt at the end of smp_unlock as well. Otherwise, the unknown conflict occurs, too.

closed 2024-08-05 by phillippZZ