enjoy-digital/litex #2029: Using the Prog USB instead of PMODB for UART on Gatemate Evalboard

Haron123, 2024-08-03:
Im currently having the issue that the Colognechip communicates over UART with the PMOD connector, but i sadly dont have any ttl to usb adapter, i tried looking through the code but nothing suggests how to switch it over to using the PROG USB for UART communication, is this currently possible to do or is this limited to just using the IO for UART.

trabucayre, 2024-08-03:
FT2232 connected to the PROG USB connector has 2 interfaces:
- the first is used to access the SPI FLASH directly
- the second is used to access the FPGA via JTAG protocol
This mean this connector can't be used for UART and a external adapter is required.

Haron123, 2024-08-03:
I see, thanks for clarifying it!

closed 2024-08-03 by Haron123

AndrewD, 2024-08-03:
@Haron123 you could maybe use valentyusb for the console like orangecrab and other boards? 
https://github.com/litex-hub/litex-boards/blob/81209b9fd10e40b4ce7f1a98a4b25d6569a78fc8/litex_boards/targets/gsd_orangecrab.py#L161

This needs 3 3v3 gpio, one resistor and being about to generate 12mhz and 48mhz clocks for usb:
https://github.com/litex-hub/litex-boards/blob/81209b9fd10e40b4ce7f1a98a4b25d6569a78fc8/litex_boards/targets/gsd_orangecrab.py#L127

trabucayre, 2024-08-04:
Indeed it's maybe a solution but:
- PMOD are connected to FPGA via level shifters: I have successfully used PMOD for DVI but I don't know if input direction will be working too
- `CC_PLL` could generates two frequencies: 
  - `freqOut` at *xx MHz* via `CLK0`, `CLK90`, `CLK180`, `CLK270`
  - `2 * freqOut` (at *2*xxx MHz*) via `CLK180_DOUB` and `CLK270_DOUB`.

Haron123, 2024-08-04:
I might try to get that working, currently i hacked an Arduino i had laying around to work as a USB-TTL converter. Thanks for the Idea