enjoy-digital/litex #2034: `litex.build.gowin` is missing a proper `__init__.py`

Mai-Lapyst, 2024-08-10:
The [`__init__.py` of the `litex.build.gowin` module](https://github.com/enjoy-digital/litex/blob/6309f30e0b3ff807249a6003730d3920bc88d0c0/litex/build/gowin/__init__.py) is miss-configured, as it contains no code at all. Looking at other modules [like the one for xilinx](https://github.com/enjoy-digital/litex/blob/6309f30e0b3ff807249a6003730d3920bc88d0c0/litex/build/xilinx/__init__.py) it can be seen that there's ment to be code in it.

The missing code leads to a problem when a programmer tries to use anything gowin related. I.e. following code will fail to be executed:
```py
from litex.build.gowin import GowinPlatform
```

While when using xilinx on the other hand it works perfectly fine:
```py
from litex.build.xilinx import XilinxPlatform
```

AndrewD, 2024-08-10:
Several boards use this, for example: 

[sipeed_tang_nano_20k](https://github.com/litex-hub/litex-boards/blob/81209b9fd10e40b4ce7f1a98a4b25d6569a78fc8/litex_boards/platforms/sipeed_tang_nano_20k.py#L10)

So the issue may be bit quite this simple. Can you build any of those existing targets?

Mai-Lapyst, 2024-08-10:
Yes if I use `from litex.build.gowin.platform import GowinPlatform` directly, it works like intended. My report is explicitly for the syntax of `from litex.build.gowin import GowinPlatform`, which is frequently used in other parts with other platforms like xilinx. One such example are the litex example projects like here: https://github.com/litex-hub/fpga_101/blob/master/lab001/base.py . It uses `from litex.build.xilinx import XilinxPlatform`, so if I want to use a gowin board in these examples instead I expect to be able to rewrite it into `from litex.build.gowin import GowinPlatform` and it should work; however this is not the case currently.

AndrewD, 2024-08-10:
Would you consider implementing implementing the required change that adds the imports to __init__, then raising a pull request?

commit 3d0fe4eb (2024-08-11): Fix litex.build.gowin's __init__.py; closes #2034
commit b279fc9f (2024-08-11): Merge pull request #2035 from Mai-Lapyst/fix_empty_initpy

Fix litex.build.gowin's __init__.py; closes #2034
closed 2024-08-11 by trabucayre