enjoy-digital/litex #2040: No longer get a .we read strobe when a CSRStatus is read if using the picorv32 soft cpu

dwalton65, 2024-08-14:
This was working previously. Yesterday I updated litex, and since then I no longer get a .we read strobe when a CSRStatus is read.
I am using the picorv32 soft cpu. If I change to the vexriscv soft cpu, then the .we read strobe works again.

I need to use the picorv32 soft cpu, as I have been unable to get litex working when using the vexriscv soft cpu and the diamond toolchain. I need to use the diamond toolchain, as I have been unable to get IDDR71B primitives working with the trellis toolchain.

[test3.gz](https://github.com/user-attachments/files/16608805/test3.gz)
"test3.gz" contains a project to test the .we read strobe. This is built by running ./litex_build

If I enter cntr on the litex_term, I get the following:
picorv32:
```
rdata = 0xaa
re_cntr = 0
we_cntr = 0
wdata_cntr = 2
```

vexriscv:
```
rdata = 0xaa
re_cntr = 0
we_cntr = 1
wdata_cntr = 2

```
we_cntr remains 0 when using the picorv32.

picorv32: litescope and gtkwave showing the point where I expect a .we strobe. main_tst_rdata_we remains low.
[image]

vexriscv: .we strobe (main_tst_rdata_we) is as expected.
[image]

dwalton65, 2024-08-15:
I reverted to the previous version of the files csr_bus.py, ahb.py and wishbone.py
```
git checkout dde9605 -- litex/soc/interconnect/csr_bus.py
git checkout b19d992 -- litex/soc/interconnect/ahb.py
git checkout fbf03ec -- litex/soc/interconnect/wishbone.py
```
Now if I enter cntr on the litex_term, I get the following:
picorv32:
```
rdata = 0xaa
re_cntr = 0
we_cntr = 1
wdata_cntr = 2

```
"we_cntr = 1" shows the .we read strobe has been seen.

picorv32: .we strobe (main_tst_rdata_we) is as expected again.
[image]

dwalton65, 2024-08-15:
Accidentally closed, so reopened.

m-byte, 2024-08-17:
This sounds like it might also have been caused by #1999. Reading through the commit, I feel like it's a breaking change that should have been better communicated before merging into master. It also broke the etherbone crossover UART.

enjoy-digital, 2024-08-19:
Hi @dwalton65,

as pointed by @m-byte, this is probably caused by #1999 that changed CSR access behavior that was previously masking some Wishbone `sel` issues on Wishbone Masters. This should be fixed with https://github.com/enjoy-digital/litex/commit/afc66fd5cfd048c5881c4d0ae0ce008874ee99f8, can you try it?

dwalton65, 2024-08-22:
> Hi @dwalton65,
> 
> as pointed by @m-byte, this is probably caused by #1999 that changed CSR access behavior that was previously masking some Wishbone `sel` issues on Wishbone Masters. This should be fixed with [afc66fd](https://github.com/enjoy-digital/litex/commit/afc66fd5cfd048c5881c4d0ae0ce008874ee99f8), can you try it?

The  .we read strobe when a CSRStatus is read appears to be working again. I tested [6d0ae25](https://github.com/enjoy-digital/litex/commit/6d0ae25d65ec60525a7440369494d588bf23095f)

enjoy-digital, 2024-08-22:
Great, thanks @m-byte for the feedback.

closed 2024-08-22 by enjoy-digital