enjoy-digital/litex #2051: Avoiding flatten hierarchy of SOC modules

ng2rdev, 2024-08-28:
Is there a way, for when generating the verilog code of a SOC for a specific board (say digilent_nexys) to not get the verilog netlist with a flatten hierarchy. 

I mean, besides usually the CPU, the module boundaries are not preserved and it is extremely difficult to debug the interaction of hardware/software/external interfaces...

I wonder if there is an easy option ( command switch, change in the litex source code) to not get a verilog flatten hierarchy and see the modules/peripheral boundaries (uarts, spis, timers, video buffers.., bus controllers/muxes and so on).

enjoy-digital, 2024-08-30:
Hi @ng2rdev,

this is not yet possible but will probably be done in the future.

ng2rdev, 2024-09-01:
Thanks for the information. Keep doing the good work!