enjoy-digital/litex #2054: Can't choose apicula as toolchain

nerfur, 2024-08-31:
While running it as 
> ./litex-boards/litex_boards/targets/sipeed_tang_nano_20k.py --toolchain=apicula --build 

getting this error 

```
INFO:SoC:Initializing ROM rom with contents (Size: 0x64e8).
INFO:SoC:Auto-Resizing ROM rom from 0x20000 to 0x64e8.
Traceback (most recent call last):
  File "/home/nerfur/FPGA/litex/./litex-boards/litex_boards/targets/sipeed_tang_nano_20k.py", line 155, in <module>
    main()
  File "/home/nerfur/FPGA/litex/./litex-boards/litex_boards/targets/sipeed_tang_nano_20k.py", line 144, in main
    builder.build(**parser.toolchain_argdict)
  File "/home/nerfur/FPGA/litex/litex/litex/soc/integration/builder.py", line 415, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nerfur/FPGA/litex/litex/litex/soc/integration/soc.py", line 1498, in build
    return self.platform.build(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nerfur/FPGA/litex/litex/litex/build/gowin/platform.py", line 47, in build
    return self.toolchain.build(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/nerfur/FPGA/litex/litex/litex/build/generic_toolchain.py", line 123, in build
    self.run_script(script)
  File "/home/nerfur/FPGA/litex/litex/litex/build/gowin/gowin.py", line 180, in run_script
    raise OSError(msg)
OSError: Unable to find Gowin toolchain, please:
- Add Gowin toolchain to your $PATH.
```

trabucayre, 2024-08-31:
Apicula toolchain is now quite fully supported by LiteX but boards are not all updated to propagate toolchain argument: this is the goal to this PR https://github.com/litex-hub/litex-boards/pull/603

nerfur, 2024-08-31:
Thanks for info!
I tried to add "toolchain" from commits.

If you are interested - apycula started running, but got 

> ERROR: Module `\ODDR' referenced in module `\sipeed_tang_nano_20k' in cell `\ODDR' is not part of the design.

Anyway I have nano_9k and nano_20k, so if you need some test running feel free to ask!