enjoy-digital/litex #2061: Vexriscv SMP interrupts not working with Zephyr

lcomino64, 2024-09-06:
Hi all,

Here is the issue described here: https://github.com/zephyrproject-rtos/zephyr/discussions/78053#discussion-7141689

I'm not entirely sure this is a LiteX issue since the LiteX BIOS works perfectly on my board, it's when I run my Zephyr program that interrupts start to malfunction and freeze the program (see issue).

I managed to run a sim of a similar configuration:
```
litex_sim --cpu-type=vexriscv_smp --cpu-count=2 --with-sdram --sdram-init ~/zephyrproject/zephyr/build/zephyr/zephyr.bin --with-rvc --with-privileged-debug --hardware-breakpoints 4 --jtag-tap --with-jtagremote
```
Where the `zephyr.bin` is the `samples/subsys/console/echo` sample from Zephyr
The program in the sim doesn't get as far but I would still like to see if the issue is present there, this is the output:
[image]

I just have no idea how to 'step through' the execution or analyse the output. Can anyone help me with this? Greatly appreciate any help.

lcomino64, 2024-09-06:
I've attached here the zephyr.bin
[zephyr.bin.gz](https://github.com/user-attachments/files/16907660/zephyr.bin.gz)

Here is my Zephyr fork:
https://github.com/lcomino64/zephyr-lcomino64

And here is a repo containing my LiteX configuration and build output:
https://github.com/lcomino64/Thesis-public/tree/main/litex

Dolu1990, 2024-09-06:
Hi,

Can you send the elf file of the compiled zephyr aswell ?
Thanks ^^

About debugging with openocd the simulated VexRiscv, you can use : 
https://github.com/enjoy-digital/litex/pull/1887

But be aware, this is kinda slow.

lcomino64, 2024-09-06:
Here: 
[zephyr.elf.gz](https://github.com/user-attachments/files/16908063/zephyr.elf.gz)

Thanks, I'll into openocd

Dolu1990, 2024-09-06:
Got a "riscv64-unknown-elf-objdump: zephyr.elf: file format not recognized" when tried to make a deassembly from the elf XD
which is very weird

Overall i just wanted to check there was no PMP related code, as vexriscv has no PMP support by default. And i know that zephyr my assume there is a pmp depending how it is configured

lcomino64, 2024-09-09:
Hey @Dolu1990,

I have the debugger on the sim up and running (turns out the sim had different csr values duh), can confirm I get the same issue with ISR1 stack maxed out. 

I'm looking through the execution now, I can see the Zephyr kernel throwing stack-related errors but I need help actually identifying the issue.

Can you please lend a hand?

I've made it the hello_world sample since the problem still occurs and there's less stuff going on.

Here's the elf and binary:
[elf_bin.tar.gz](https://github.com/user-attachments/files/16929956/elf_bin.tar.gz)

And here are the zephyr files for the sim in case you want to build for yourself. Just unzip to your zephyr directory.
[zephyr_changes_sim.tar.gz](https://github.com/user-attachments/files/16929959/zephyr_changes_sim.tar.gz)