enjoy-digital/litex #2062: TIMER0 Interrupt

gcourchinoux, 2024-09-06:
Hi everyone, 

I try to get the interrupt from the timer0 on vexriscv SOC. 

I have read the documentation and on it there is this sentence : "For both modes, **the CPU can be advertised by an IRQ** that the duration/period has elapsed. (The CPU can also do software polling with update_value and value to know the elapsed duration)"

I searched on the documentation  but I didn't found how to set the interrupt in Periodic mode and One Shot mode. 

Does anyone know how to activate the IRQ? 

Thank in advance, 

Gaspard

Haron123, 2024-09-08:
Is the Question how to specifically use an Interrupt or rather how to setup the Timer to be in a specific mode?
Or do you just need a Guide for it both?

gcourchinoux, 2024-09-14:
Hi, 
If you have a guide for both it will be with pleasure !

Haron123, 2024-09-15:
ill try my best to explain how ive accomplished to trigger the interrupt with the timer:

### Activating the Interrupt

First, obviously, you need a timer with the Irq enabled (Target file):
```
self.timer1 = Timer()
self.irq.add("timer1", use_loc_if_exists=True)
```

To enable it inside of my code, i use the following initialization routine
```C++
	irq_setie(0);

	timer1_ev_pending_write(timer1_ev_pending_read());
	timer1_ev_enable_write(1);
	irq_attach(TIMER1_INTERRUPT, uart0Isr);
	irq_setmask(irq_getmask() | 1 << TIMER1_INTERRUPT);

	irq_setie(1);
```
You will have to include <irq.h> for all irq related function calls and possibly the <generated/csr.h> aswell to get the function, which write into the Register of the timer.

### Using the Timer

For both modes the Documentation on the timer states this

To use the Timer in One-Shot mode, the user needs to:

    Disable the timer

    Set the load register to the expected duration

    (Re-)Enable the Timer

To use the Timer in Periodic mode, the user needs to:

    Disable the Timer

    Set the load register to 0

    Set the reload register to the expected period

    Enable the Timer

You should be able to just call the right registers for the Timer, incase you need help you can check the <generated/csr.h> file in ur build directory and also use the --doc flag when running the target file to get a html representation of the Documentation in the build/doc directory.