enjoy-digital/litex #2084: Simulated SoC communicate through UART

Leopard777, 2024-09-28:
I had a look at the wiki (https://github.com/enjoy-digital/litex/wiki/SoC-Simulator and https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC) and I had the question whether or not its possible to simulate an SoC with UART and communicate with it through UART? The reason I’m asking is because the firmware I’m creating accepts commands via UART and I would like to make use of simulation for development.

P.S. The project looks really cool!

AndrewD, 2024-09-28:
You can interact with the bios, or a running firmware, via the terminal when running the simulator.

enjoy-digital, 2024-10-15:
Hi @Leopard777,

do you want the simulator to create a virtual UART and just use regular terminal tools to interact with it? If so, this is not currently possible with upstream code, but I probably have some old code to do so and would try to re-integrate it.

Leopard777, 2024-10-15:
Hi,

Apologies for the delay @AndrewD. @enjoy-digital yeah, I think that’s exactly what I need. The simulator creates a virtual UART for the SoC and some client side code can send/receive data by means of a FIFO or a socket, … . But I have no idea have feasible this is.

closed 2026-06-16 by enjoy-digital