enjoy-digital/litex #2091: Generated Verilog project can't work

snowprogrammer, 2024-10-08:
1. I use this command to generate a Verilog project with wishbone bus: `python3 -m litex_boards.targets.digilent_basys3 --integrated-rom-init=test.bin --integrated-main-ram-size=0x2000 --build`, then I test the Verilog file, and it works well in simulation in Vivado.
2. Then I change to AXI bus: `python3 -m litex_boards.targets.digilent_basys3 --integrated-rom-init=test.bin --integrated-main-ram-size=0x2000 --bus-standard axi-lite --build`,   while I launch the simulation, there is an error : **FATAL_ERROR: Iteration limit 10000 is reached. Possible zero delay oscillation detected where simulation time can not advance. Please check your source code. Note that the iteration limit can be changed using switch -maxdeltaid. Time: 1493 ns  Iteration: 10000**
3. It seems there is an error in the architecture, I wonder it is because the card basys3 is not suitable with AXI? Or this is an error in my command?

enjoy-digital, 2024-10-15:
Hi @snowprogrammer,

are you able to build the design with `--build`? That's possible a part of the code create a combinatorial loop that would need to be examined/removed. Can you look if you have a warning regarding this while building with Vivado?

snowprogrammer, 2024-10-15:
> Hi @snowprogrammer,
> 
> are you able to build the design with `--build`? That's possible a part of the code create a combinatorial loop that would need to be examined/removed. Can you look if you have a warning regarding this while building with Vivado?

Hello. I can build the design with `--build` and the implementation with demo.bin can work, only the Verilog file is not correctly generated. In Vivado I use` xsim -tp -tl `and find 6 loops in the architecture. Because I'm not familiar with this architecture, I haven't fixed the problem.

murattokez, 2025-03-14:
Has anyone had this problem and solved it?

snowprogrammer, 2025-03-14:
> Has anyone had this problem and solved it?

I manually correct the code in Verilog and works

murattokez, 2025-03-14:
I think the problem is in the ready-valid handshake? Any chance to share the source of the problem and how you solved it?

snowprogrammer, 2025-03-14:
> I think the problem is in the ready-valid handshake? Any chance to share the source of the problem and how you solved it?

In the code, some "if condition" are badly wrote, I change them and it works.