enjoy-digital/litex #2107: Cant have multiple Instances of Custom VHDL core

Haron123, 2024-10-26:
The **Error Message** i get is

> own_periphs/uart/UART_1.v:1: ERROR: Re-definition of module `$abstract\uart_clk_div_17_1'!

When i try to instantiate multiple of the same cores, in the **Target file** i defined them like this:
```
self.uart_logging = CustomUart(self.platform, platform.request("uart_logging"), sys_clk_freq, 115200)
self.irq.add("uart_logging", use_loc_if_exists=True)

self.uart_ice40 = CustomUart(self.platform, platform.request("uart_ice40"), sys_clk_freq, 115200)
self.irq.add("uart_ice40", use_loc_if_exists=True)
```

The **VHDL To Verilog** conversion looks like this
```python
# Convert to Verilog
self.vhd2v_converter = VHD2VConverter(platform,
	top_entity = "UART",
	build_dir = os.path.abspath(os.path.dirname(__file__)),
	params = self.parameters,
	force_convert = True,
	add_instance = True,
	work_package = "uart",
)

self.sources_path = os.path.abspath(os.path.dirname(__file__)) + "/comp/";

self.sources = [
	# uart.vhd
	"uart.vhd",

	# comp
	"uart_debouncer.vhd",
	"uart_parity.vhd",
	"uart_rx.vhd",
	"uart_tx.vhd",
	"uart_clk_div.vhd",
]

self.vhd2v_converter.add_sources(self.sources_path, *self.sources)
```

Does anyone know what could cause it? 

### Debug Attempts:
I know that it works when i only use one VHDL File, so it seems to be an issue with converting multiple VHDL Files. I looked into the VHD2V Code and apparently it only renames the TOP Entity, i didnt manage to change the code to rename it when theres multiple subentities.

Thanks in advance for any attempts to help.

Haron123, 2024-11-02:
Im still stuck on this, does anybody know what could cause it ?

pepijndevos, 2025-03-24:
I'm facing the same problem.

The cause is that the VHDL converter only renames the top instance when there are multiple instances. Any other modules will have collisions.

https://github.com/enjoy-digital/litex/blob/b1eb23bec2ae2c9730b71a740d62d015579fa30c/litex/build/vhd2v_converter.py#L204

closed 2025-04-28 by trabucayre