enjoy-digital/litex #2110: litex_sim hangs when adding a call to a function in libliteeth

jersey99, 2024-10-30:
To reproduce:
```
litex_sim.py --with-ethernet --local-ip 192.168.2.50 --remote-ip 192.168.2.100 --integrated-main-ram-size=0x10000 --cpu-type=vexriscv --no-compile-gateware
BUILD_DIR=../build/sim make -C demo clean all
litex_sim.py --with-ethernet --local-ip 192.168.2.50 --remote-ip 192.168.2.100 --integrated-main-ram-size=0x10000 --cpu-type=vexriscv --ram-init=demo/demo.bin
```
This needs to be run with the patch below, on main.c

```
10a11
> #include <libliteeth/udp.h>
204a206,209
> #ifdef CSR_ETHMAC_BASE
>     printf("foo\n");
>     eth_init();
> #endif
```

trabucayre, 2024-12-10:
Hi,
I'm able to reproduce your issue and in fact its also reproducible with a physical target.
I have tried to use openocd + gdb to see why the CPU hangs.

To do that I have added a breakpoint on `printf` just before `eth_init` call:
- with `eth_init` commented: execution stop at this specific line, ie to code is fine
- with `eth_init` un-commented: CPU never execute this instruction and remains somewhere between `trap_entry` and `isr`

kscz, 2025-01-15:
I accidentally re-reported this issue in https://github.com/enjoy-digital/litex/issues/2157 - I am eager to hear if there is anything I can do to move this problem along.