enjoy-digital/litex #2112: Video-terminal useable funtion?

2jack5657, 2024-11-04:
I just finished building the SoC, and I added the option `--with-video-terminal` when compiling. When I booted the bios after the SoC was successfully burned, HDMI did work and output the same console interface as the serial port, but what I want to ask is, doesn't video-terminal have related .c or .h files to give functions for calling?

enjoy-digital, 2025-02-18:
Hi @2jack5657,

this video terminal is directly handled in FPGA logic, so does not have corresponding .c/.h files. For a Video Terminal directly controlled from the CPU, you can look at: https://github.com/litex-hub/litevideo/tree/master/litevideo/terminal

2jack5657, 2025-02-20:
ok I got it, thank you 4 u reply, have a nice day!