enjoy-digital/litex #2121: CVA6 + Nexys4 DDR - RAM access issues

juanschroeder, 2024-11-08:
Hi there,

I'm trying to test the CVA6 in my Nexys4 DDR board. Unfortunately I don't have a Genesys2 board.

Basic simulation seems to work without issues, but running on the FPGA board gives errors startup: 'Memory initialization failed'.

Console seems to work fine, but again, both 'sdram_test' and 'mem_test' command seems to fail ('KO' results)
I've built the bitstream for 75MHz (default), 50MHz, 25MHz and 15MHz and the issue still happens.

When running for 'smaller' sizes, mem_test seems to work well most of the time when run consecutively (except always from the first time):

`litex> mem_test 0x40000000 0x1000
Memtest at 0x40000000 (4.0KiB)...
  Write: 0x40000000-0x40001000 4.0KiB   
   Read: 0x40000000-0x40001000 4.0KiB   
Memtest OK
`

Can you suggest a good way to debug this issue? Is there something I can check in simulation?
Can I easily add features to the 'bios' running to get more info?

Thanks!

FlyGoat, 2024-12-27:
Can reproduce the issue on alibaba_vu13p, that’s wired.

rickliu2000, 2024-12-27:
Here's some console output of the cva6 on alibaba vu13p
Also, the memory latency test doesn't work.
[image]
[image]

https://pastebin.com/xMyecG12