enjoy-digital/litex #2126: No compatibility with Efinity 2024.2.294

m-byte, 2024-11-20:
There are multiple changes in 2024.2 that seem to break compatibility:
1. the pt/db/device_*.xml files are gone. This might be intentional, however, I just copied them over from the previous version for myself, which seems to help (will investigate further at some later point in time)
2. efinity seems to prepend work_syn to the name of the vdb file.

I will update this as I go...

enjoy-digital, 2024-12-12:
Thanks @m-byte, we can probably close this since your PR has been merged.

m-byte, 2024-12-12:
@enjoy-digital not quite, as I only took care of the secind point above.

enjoy-digital, 2024-12-12:
@m-byte: OK thanks, that's re-opened then.

jwise, 2025-01-15:
Good idea on copying over the xml file from the previous version :) I looked briefly to see if the name had changed, and it appears that not just the name has changed but also the structural content.

closed 2025-02-20 by m-byte

jwise, 2025-02-21:
Thanks so much, @m-byte!