enjoy-digital/litex #2136: Memory initialization files

merveyubogluu, 2024-12-11:
Hi, I wanted to understand how Litex actually creates the memory files using Migen. I wanted to compile my own C codes and be able to use the created binary. I actually used the --integrated_rom_init method for instructions but i couldnt find a way for extracting the data memory information from the binary and did not understand how Litex does it. 

Does anyone know something about this?

enjoy-digital, 2024-12-12:
Hi @merveyubogluu,

When you use `--integrated-rom-init` and pass a binary, LiteX automatically invokes `get_mem_data()` on that binary to produce the initialization arrays. These arrays are then used directly to initialize the on-chip ROM, as can be seen here:

https://github.com/enjoy-digital/litex/blob/a32096e80602e4949565c721ca6d620d44270a60/litex/soc/integration/soc_core.py#L163

```python3
        # ROM.
        # Initialize ROM from binary file when provided.
        if isinstance(integrated_rom_init, str):
            integrated_rom_init = get_mem_data(integrated_rom_init,
                endianness = "little", # FIXME: Depends on CPU.
                data_width = bus_data_width
            )
            integrated_rom_size = 4*len(integrated_rom_init)
```

Otherwise, more generally: LiteX uses get_mem_data() to convert a given binary or JSON memory image into Python arrays that initialize its internal memories at build time. These arrays are then passed as the init parameter to Migen’s Memory modules, ensuring that when the design is synthesized and loaded onto the FPGA, the memory blocks already contain the specified instructions or data.

merveyubogluu, 2024-12-12:
Hi thanks for the quick response. I understood the part what get_mem_data function does however it actually generates the instructions from the binary only. I was wondering how other memories are generated? Cause when I create an SoC without using the --integrated-rom-init, it generates a memory file. When I include a binary, it still generates the other memory but with a ROM (instruction memory) also. My question was how Litex build that memory (in this case i think its data memory) not the instruction memory.