enjoy-digital/litex #2138: efinix_trion_t20_mipi_dev_kit SPI flash and litex demo issues

Kanken6174, 2024-12-13:
---
test environment : arch linux 6.12.4-arch1-1; efinix toolchain efinity version 2023.2
target/example : efinix T20 mipi DK, efinix_trion_t20_mipi_dev_kit.py, default litex_bare_metal_demo
---

Hi all,

I have been trying to get litex to work on an efinix T20 mipi DK, which already has a board script in litex-boards. There are two main issues:

- The spi chip doesn't seem to be flashable from the litex "load" argument when calling `python3 litex-boards/litex_boards/targets/efinix_trion_t20_mipi_dev_kit.py --build --with-spi-flash --load`. It doesn't actually do anything to flash the spi chip, i've confirmed this through my logic analyzer. All it does is send the bit stream to RAM and it will be erased next reset. Flashing the .hex file produced by litex from the official efinix programmer does cause the intended behavior and the spi rom is flashed properly.

- Loading the demo software onto the system in this ram state, or even flashed to rom through the efinix programmer, causes a data corruption error:
it does build:
```yaml
➜  litex litex_bare_metal_demo --build-path=build/efinix_trion_t20_mipi_dev_kit         
cp: -r not specified; omitting directory '/home/kanken/opt/litex/litex/litex/soc/software/demo/build'
cp: -r not specified; omitting directory '/home/kanken/opt/litex/litex/litex/soc/software/demo/__pycache__'
 CC       donut.o
 CC       helloc.o
 CC       main.o
 CC       demo.elf
/opt/riscv64-gnu-toolchain-elf-bin/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld:linker.ld:23: warning: memory region `main_ram' not declared
/opt/riscv64-gnu-toolchain-elf-bin/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: demo.elf has a LOAD segment with RWX permissions
chmod -x demo.elf
 OBJCOPY  demo.bin
chmod -x demo.bin
```
then, resetting the board which has had litex flashed to spi rom through the efinix programmer previously:
```yaml
➜  litex litex_term /dev/ttyUSB0 --kernel=demo.bin                                      

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 13 2024 15:50:14
 BIOS CRC passed (ddceaec9)

 LiteX git sha1: f63d4a833

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		wishbone 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
FLASH:		4.0MiB

--========== Initialization ============--

Initializing w25q32jv SPI Flash @0x00400000...
SPI Flash clk configured to 25 MHz
Memspeed at 0x400000 (Sequential, 4.0KiB)...
   Read speed: 2.6MiB/s
Memspeed at 0x400000 (Random, 4.0KiB)...
   Read speed: 1.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x40000000 (7684 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload to device failed due to data corruption (CRC error)
```
The actual application binary fails to be sent over due to this error. Is this a known issue with this board? I didn't see that many posts about the T20.

(i didn't open two different issues because i have a feeling both of these problems are linked, most likely to the way litex handles the spi rom in this situation).

I'll keep looking around to see if i find a solution, but feel free to drop a hint if you have an idea.

AndrewD, 2024-12-13:
`--flash` is to write to flash, `--load` is load to ram.

There are known compatibility issues with efinity 2024.2 and litex, so maybe roll back to 2024.1 for now.

Kanken6174, 2024-12-13:
> `--flash` is to write to flash, `--load` is load to ram.
> 
> There are known compatibility issues with efinity 2024.2 and litex, so maybe roll back to 2024.1 for now.

That's one of the first commands i had attempted but it doesn't seem to work if called with this specific target:

```yaml
➜  litex python3 litex-boards/litex_boards/targets/efinix_trion_t20_mipi_dev_kit.py --build --with-spi-flash --flash       
usage: {'description': 'LiteX SoC on Efinix Trion T20 MIPI Dev Kit.'} [-h] [--toolchain {efinity}] [--build] [--load]
                                                                      [--log-filename LOG_FILENAME]
                                                                      [--log-level LOG_LEVEL]
                                                                      [--sys-clk-freq SYS_CLK_FREQ] [--with-spi-flash]
                                                                      [--output-dir OUTPUT_DIR]
                                                                      [--gateware-dir GATEWARE_DIR]
                                                                      [--software-dir SOFTWARE_DIR]
                                                                      [--include-dir INCLUDE_DIR]
                                                                      [--generated-dir GENERATED_DIR]
                                                                      [--build-backend BUILD_BACKEND] [--no-compile]
                                                                      [--no-compile-software] [--no-compile-gateware]
                                                                      [--soc-csv SOC_CSV] [--soc-json SOC_JSON]
                                                                      [--soc-svd SOC_SVD] [--memory-x MEMORY_X] [--doc]
                                                                      [--bios-lto]
                                                                      [--bios-format {integer,float,double}]
                                                                      [--bios-console {full,no-history,no-autocomplete,lite,disable}]
                                                                      [--bus-standard BUS_STANDARD]
                                                                      [--bus-data-width BUS_DATA_WIDTH]
                                                                      [--bus-address-width BUS_ADDRESS_WIDTH]
                                                                      [--bus-timeout BUS_TIMEOUT] [--bus-bursting]
                                                                      [--bus-interconnect BUS_INTERCONNECT]
                                                                      [--cpu-type CPU_TYPE] [--cpu-variant CPU_VARIANT]
                                                                      [--cpu-reset-address CPU_RESET_ADDRESS]
                                                                      [--cpu-cfu CPU_CFU] [--no-ctrl]
                                                                      [--integrated-rom-size INTEGRATED_ROM_SIZE]
                                                                      [--integrated-rom-init INTEGRATED_ROM_INIT]
                                                                      [--integrated-sram-size INTEGRATED_SRAM_SIZE]
                                                                      [--integrated-main-ram-size INTEGRATED_MAIN_RAM_SIZE]
                                                                      [--csr-data-width CSR_DATA_WIDTH]
                                                                      [--csr-address-width CSR_ADDRESS_WIDTH]
                                                                      [--csr-paging CSR_PAGING]
                                                                      [--csr-ordering CSR_ORDERING] [--ident IDENT]
                                                                      [--no-ident-version] [--no-uart]
                                                                      [--uart-name UART_NAME]
                                                                      [--uart-baudrate UART_BAUDRATE]
                                                                      [--uart-fifo-depth UART_FIFO_DEPTH]
                                                                      [--with-uartbone] [--with-jtagbone]
                                                                      [--jtagbone-chain JTAGBONE_CHAIN] [--no-timer]
                                                                      [--timer-uptime] [--with-watchdog]
                                                                      [--watchdog-width WATCHDOG_WIDTH]
                                                                      [--watchdog-reset-delay WATCHDOG_RESET_DELAY]
                                                                      [--l2-size L2_SIZE]
                                                                      [--synth-mode {speed,area,area2}]
                                                                      [--infer-clk-enable {0,1,2,3,4}]
                                                                      [--infer-sync-set-reset {0,1}]
                                                                      [--bram-output-regs-packing {0,1}]
                                                                      [--retiming {0,1,2}] [--seq-opt {0,1}]
                                                                      [--mult-input-regs-packing {0,1}]
                                                                      [--mult-output-regs-packing {0,1}]
{'description': 'LiteX SoC on Efinix Trion T20 MIPI Dev Kit.'}: error: unrecognized arguments: --flash
```
It works with other targets such as arty though.

Kanken6174, 2024-12-13:
hmm this doesn't make much sense... the load command appears to work for the BGA256 T20 devkit, but not for the T20 MIPI one, or the T375 one... this seems like a target misconfiguration, i'll have to find what changes between the two...

Kanken6174, 2024-12-13:
okay... that's a very simple fix, just go to the issue-causing .py target and add this line:
```py
    from litex.build.parser import LiteXArgumentParser
    parser = LiteXArgumentParser(platform=efinix_trion_t20_mipi_dev_kit.Platform, description="LiteX SoC on Efinix Trion T20 MIPI Dev Kit.") #that one here
    parser.add_target_argument("--flash",          action="store_true",             help="Flash bitstream.")
    parser.add_target_argument("--sys-clk-freq",   default=100e6, type=float, help="System clock frequency.")
   ```
   it fixes it
   
   That's one issue down, thanks for somehow pointing me in the right direction... the corruption one still stands

AndrewD, 2024-12-13:
> hmm this doesn't make much sense... the `--flash` command appears to work for the BGA256 T20 devkit, but not for the T20 MIPI one, or the T375 one... this seems like a target misconfiguration, i'll have to find what changes between the two...

Compare the platform file for the targets. Sounds like flash hadn't been defined for these platforms: should be a simple change.

Kanken6174, 2024-12-13:
Actually that's not the only thing that needs changing i think, while now it accepts the argument it doesn't actually call any flash logic, i'll dig deeper to see what changes between the targets

Kanken6174, 2024-12-13:
One missing part is this as well:
```py
    if args.flash:
        from litex.build.openfpgaloader import OpenFPGALoader
        prog = OpenFPGALoader("trion_t120_bga576")
        prog.flash(0, builder.get_bitstream_filename(mode="flash", ext=".hex")) # FIXME
```
Now it does actually work 👍🏻 
```yaml
Detected: Winbond W25Q32 64 sectors size: 32Mb                                                                        │
00000000 00000000 00000000 00                                                                                         │
Erasing: [==================================================] 100.00%                                                 │
Done                                                                                                                  │
Writing: [==================================================] 100.00%                                                 │
Done                                                                                                                  │
Reset DONE
```

AndrewD, 2024-12-13:
It would be great if you can submit a pr with your tested fix.

Kanken6174, 2024-12-13:
yeah i'll do that once i've validated it

Kanken6174, 2024-12-13:
There, PR is open. Though now that corruption issue remains, i don't have a real lead on what could be causing it for now:
```yaml
➜  litex litex_term /dev/ttyUSB0 --kernel=demo.bin                               

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Dec 13 2024 19:47:00
 BIOS CRC passed (7a48e9fe)

 LiteX git sha1: f63d4a833

--=============== SoC ==================--
CPU:            VexRiscv @ 100MHz
BUS:            wishbone 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128.0KiB
SRAM:           8.0KiB
FLASH:          4.0MiB

--========== Initialization ============--

Initializing w25q32jv SPI Flash @0x00400000...
SPI Flash clk configured to 50 MHz
Memspeed at 0x400000 (Sequential, 4.0KiB)...
   Read speed: 5.1MiB/s
Memspeed at 0x400000 (Random, 4.0KiB)...
   Read speed: 2.2MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x40000000 (7684 bytes)...
[LITEX-TERM] Upload calibration... (inter-frame: 10.00us, length: 64)
[LITEX-TERM] Upload to device failed due to data corruption (CRC error)
```

Given that the original BIOS runs fine the soc setup should be ok, it could be a binary file issue (wrong arch...) but given that it's generated from the demo command it seems like a strange issue, i'll have to dig deeper into how the demo script builds this.
```yaml
➜  litex litex_bare_metal_demo --build-path=build/efinix_trion_t20_mipi_dev_kit                                                             
cp: -r not specified; omitting directory '/home/kanken/opt/litex/litex/litex/soc/software/demo/__pycache__'
cp: -r not specified; omitting directory '/home/kanken/opt/litex/litex/litex/soc/software/demo/build'
 CC       donut.o
 CC       helloc.o
 CC       main.o
 CC       demo.elf
/opt/riscv64-gnu-toolchain-elf-bin/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld:linker.ld:23: warning: memory region `main_ram' not declared
/opt/riscv64-gnu-toolchain-elf-bin/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: demo.elf has a LOAD segment with RWX permissions
chmod -x demo.elf
 OBJCOPY  demo.bin
chmod -x demo.bin
```

Kanken6174, 2024-12-13:
ok sending it over in `--safe` mode appears to work, though very slow:
```yaml
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading demo.bin to 0x40000000 (7684 bytes)...
[LITEX-TERM] Upload complete (0.1KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

Kanken6174, 2024-12-13:
I'll close this PR for now since the corruption issue seems to be non-blocking and can be avoided with --safe, it's more of a litex_term issue than anything else.

closed 2024-12-13 by Kanken6174