enjoy-digital/litex #2164: Litex last (master) not build project

faa00, 2025-01-20:
Last Litex (branch master) not build project https://github.com/enjoy-digital/HectaScope.git.

`WARNING: [Synth 8-6901] identifier 'm_axis_cq_tlast_lat' is used before its declaration [/home/faa/Project/Litex/litepcie/litepcie/phy/xilinx_usp/m_axis_cq_adapt_256b.v:54]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:14668]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29191]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29192]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29195]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29196]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29199]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29200]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29203]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29204]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29445]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29446]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29449]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29450]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29453]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29454]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29457]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29458]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29699]
ERROR: [Synth 8-2716] syntax error near '[' [/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v:29700]
ERROR: [Synth 8-12188] Failed to read verilog '/home/faa/HectaScope/build/alinx_axau15/gateware/alinx_axau15.v'
`

Vivado 2024.2.

Litex (tag 2024.12) before 13/01/2025 working - project HectaScope build.

enjoy-digital, 2025-01-20:
Thanks for the feedback, it should be fixed with https://github.com/enjoy-digital/litex/commit/666c9b430984febc0d8334210e78b2a976716ea5.

faa00, 2025-01-20:
Working. 10x.

enjoy-digital, 2025-01-20:
Good, thanks for the feedback.

closed 2025-01-20 by enjoy-digital