enjoy-digital/litex #2169: NaxRISCV Bare Metal Demo not working

BMorgan1296, 2025-01-26:
Hi,

I am experimenting with different cores and found that the bare metal demo does not work for NaxRISCV under simulation or FPGA.

I am able to boot, but only get the "Liftoff!" printout and nothing else once it jumps into the demo.bin. This is due to the processor stalling on the interrupt handling routines specified in demo's main.c:

```C
int main(void)
{
// #ifdef CONFIG_CPU_HAS_INTERRUPT      //       <-- If I comment these out I get the donut :)
//	 irq_setmask(0);
//	 irq_setie(1);
// #endif
	uart_init();

	donut_cmd();

	help();
	prompt();

	while(1) {
		console_service();
	}

	return 0;
}

```

Having spent a bit of time figuring out why, this seems related to #1754 and #1584, where the processor did not have its interrupt handling defined.

However, I checked `/soc/cpu/naxriscv/irq.h` and the interrupt handling is defined there, which I assume is used to tell the core when I type into the terminal and press enter etc. I am not familiar with how PLIC works to start debugging it, so I figured I would create an issue first.

Cheers.

BMorgan1296, 2025-01-26:
Ok I have figured it out.

The NaxRISCV is booting into user mode, and for comparison, the VexRISCV is in machine mode.

Mstatus for the NaxRISCV is 0x2000, i.e. MPP field which is bits 11,12 = 0x00 == User mode.

Mstatus for the NaxRISCV is 0x1800, i.e. MPP field which is bits 11,12 = 0x11 == Machine mode.

Just figuring out how to change this now :)

Dolu1990, 2025-01-27:
Hi ^^

> The NaxRISCV is booting into user mode

NaxRiscv boot in machine mode.
The mstatus.mpp field doesn't indicate the current privilege mode, the the privilege mode which will be taken on the next mret instruction.

If do baremetal stuff only, you never need to change the mpp value, because once the CPU enter an interrupt, the MPP field is changed to the privilege the CPU was in (machine).

> And now all is working. Is this the appropriate method to do enable interrupts

This seems ok to me ^^

BMorgan1296, 2025-01-27:
Ok I’ll check other CPUs for this same issue first and make a PR to fix this so the bare metal demo works.

Dolu1990, 2025-01-29:
> One question though, why is it halting on irq_setie(1);?

It shouldn't, it is like, as you said, if NaxRiscv wasn't in machine mode anymore.
A simulation wave can easily show the issue ^^
Can you send me all the modified files + command lines you used ?

BMorgan1296, 2025-01-29:
I made a PR #2170 showing what I changed (a single character in `litex/soc/cores/cpu/naxriscv/crt0.S` lol).
Commands I ran were (after Litex installation):
```
litex_sim --integrated-main-ram-size=0x10000 --cpu-type=naxriscv  --no-compile-gateware
litex_bare_metal_demo --build-path=build/sim/
litex_sim --integrated-main-ram-size=0x10000 --cpu-type=naxriscv --ram-init=demo.bin
```

Dolu1990, 2025-01-29:
Thanks ^^
That is weird, on my side, all is going well : 

```
<DUMP ON>
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan 29 2025 13:05:03
 BIOS CRC passed (09980d8b)

 LiteX git sha1: c213c4274

--=============== SoC ==================--
CPU:		NaxRiscv 32-bit @ 1MHz
BUS:		wishbone 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
MAIN-RAM:	64.0KiB

--========== Initialization ============--

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40000000

--============= Liftoff! ===============--

LiteX minimal demo app built Jan 29 2025 13:01:43

Available commands:
help               - Show this command
reboot             - Reboot CPU
donut              - Spinning Donut demo
helloc             - Hello C
litex-demo-app> donut   
litex-demo-app> donut
Donut demo...

                                                                               
                                                                               
                                                                               
                                 $@@$$@@$$$@@$                                 
                             $$$$$###########$$$$$                             
                           $####**!!!!!!!!!!!**###$$                           
                         ####**!===============!**####                         
                       *##**!!!===;::::::::;;===!!!*###*                       
                      !****!!===;:~--,,.,,-~~:;===!!*****                      
                     =****!!==;;:-,........,--:;;=!!!****=                     
                     !***!!!==;:--...........-~:;=!!!!***!                     
                     !***!!!=;;:-,..       ..,-:;==!!!**!!                     
                    ;!!**!!!==;:-,.         ~;==!!!!**!!!=:                    
                     =!!!****!*!!!==       =!!********!!!=                     
                     ;=!!!*****#*#####***########****!!==;                     
                     -;=!!****####$$$$$@@@$$$$###****!!=;-                     
                      ~:=!!****###$$$$@@@@$$$$###**!!==;-                      
                       ,:;=!!**####$$$$$$$$$####**!!=;:-                       
                         -:;=!!!**###########**!!!=;:-                         
                           -:;;==!!!*******!!!==;::-                           
                             .-~::;;=======;;::~-.                             
                                 ..,,-----,,,.                                 
```

- I checked in the waveform, i can see the interrupts going just well
- You tried with upstream litex right ?
- You may have a old NaxRiscv netlist, to be sure, delete pythondata-cpu-naxriscv/pythondata_cpu_naxriscv/verilog/NaxRiscvLitex_*.v

Hmm, so that is weird i can't reproduce the issue.
I would say the way to move forward with this, is for me to get your simulation waveform.
For instance, i used : 
litex_sim --integrated-main-ram-size=0x10000 --cpu-type=naxriscv --ram-init=demo.bin --trace --trace-fst --trace-start=575000000000 --trace-end=590000000000
(the commits done by the CPU can be visualized by naxes_0_thread_core/commit_pc_X  and commit_mask

I'm happy to take a look into your wave.fst ^^

BMorgan1296, 2025-01-29:
Hi, thanks for that. I’ll check my repo is all updated and get you those waveforms tomorrow.

Dolu1990, 2025-01-29:
Thanks ^^

I'm using : Verilator 4.218 2022-01-17 rev v4.218

closed 2025-01-30 by BMorgan1296