enjoy-digital/litex #2173: ERROR: [DRC PDRC-43] when adding Alinx7203 board in litex-boards

TarikHamedovic, 2025-02-03:
Hello

I am trying to add a `platform` and `target` file for the `alinx7203` board on the basis of the `alinx7010` board that is already supported with LiteX but I ran into a problem when running the build command:
```
./alinx_ax7203.py --build
```

Where I got this error:
```
ERROR: [DRC PDRC-43] PLL_adv_ClkFrequency_div_no_dclk: The computed value 450.000 MHz (CLKIN1_PERIOD, net main_crg_clkin) for the VCO operating frequency of the PLLE2_ADV site PLLE2_ADV_X1Y0 (cell PLLE2_ADV) falls outside the operating range of the PLL VCO frequency for this device (800.000 - 1866.000 MHz). The computed value is (CLKFBOUT_MULT_F * 1000 / (CLKINx_PERIOD * DIVCLK_DIVIDE)). Please adjust either the input period CLKINx_PERIOD (20.000000), multiplication factor CLKFBOUT_MULT_F (9) or the division factor DIVCLK_DIVIDE (1), in order to achieve a VCO frequency within the rated operating range for this device.

```

The platform file is given below:
```
#
# This file is part of LiteX-Boards.
#
# Copyright (c) 2022 Yonggang Liu <ggang.liu@gmail.com>
# SPDX-License-Identifier: BSD-2-Clause

from litex.build.generic_platform import *
from litex.build.xilinx import Xilinx7SeriesPlatform, VivadoProgrammer

# IOs ----------------------------------------------------------------------------------------------

_io = [
    # Clk / Rst
    ("clk200",    0, Pins("R4"), IOStandard("LVCMOS33")),

    # Leds
    ("user_led", 0, Pins("B13"),  IOStandard("LVCMOS33")),
    ("user_led", 1, Pins("C13"),  IOStandard("LVCMOS33")),
    ("user_led", 2, Pins("D14"),  IOStandard("LVCMOS33")),
    ("user_led", 3, Pins("D15"),  IOStandard("LVCMOS33")),

    # Buttons
    ("user_btn", 0, Pins("T6"), IOStandard("LVCMOS33")),

    # Serial
    ("serial", 0,
        Subsignal("tx", Pins("N15"), IOStandard("LVCMOS33")),
        Subsignal("rx", Pins("P20"), IOStandard("LVCMOS33")),
    ),
]

# Connectors ---------------------------------------------------------------------------------------

_connectors = [
    ("pmodb",    "B12 B12 C12"),
    ("pmodhdmi", "N18 P19 V20 W20 T20 U20 N20 P20 R18 R16 Y18 Y19 V16"),
    ("pmodj10",  "W19 W18 R14 P14 Y17 Y16 W15 V15 Y14 W14 P18 N17 U15 U14 P16 P15 U17 T16 V18 V17 T15 T14 V13 U13 W13 V12 U12 T12 T10 T11 A20 B19 B20 C20"),
    ("pmodj11",  "F17 F16 F20 F19 G20 G19 H18 J18 L20 L19 M20 M19 K18 K17 J19 K19 H20 J20 L17 L16 M18 M17 D20 D19 E19 E18 G18 G17 H17 H16 G15 H15 J14 K14"),
]

# PMODS --------------------------------------------------------------------------------------------

_usb_uart_pmod_io = [
    # USB-UART PMOD on JB:
    # - https://store.digilentinc.com/pmod-usbuart-usb-to-uart-interface/
    ("serial", 0,
        Subsignal("tx", Pins("pmodj10:0")),
        Subsignal("rx", Pins("pmodj10:1")),
        IOStandard("LVCMOS33")
    ),
]

# Platform -----------------------------------------------------------------------------------------

class Platform(Xilinx7SeriesPlatform):
    default_clk_name   = "clk200"
    default_clk_period = 2e9/100e6

    def __init__(self):
        Xilinx7SeriesPlatform.__init__(self, "xc7a200tfbg484-2", _io, _connectors, toolchain="vivado")
        #self.add_extension(_ps7_io)
        #self.add_extension(_usb_uart_pmod_io)

    def create_programmer(self):
        return VivadoProgrammer()

    def do_finalize(self, fragment):
        Xilinx7SeriesPlatform.do_finalize(self, fragment)
        self.add_period_constraint(self.lookup_request("clk200", loose=True), 2e9/100e6)

```

And the target file is also given below:

```
#!/usr/bin/env python3

#
# This file is part of LiteX-Boards.
#
# Copyright (c) 2022 Yonggang Liu <ggang.liu@gmail.com>,
# SPDX-License-Identifier: BSD-2-Clause

from migen import *

from litex.gen import *

from litex_boards.platforms import alinx_ax7203

from litex.soc.interconnect import axi
from litex.soc.interconnect import wishbone

from litex.soc.cores.clock import *
from litex.soc.integration.soc_core import *
from litex.soc.integration.builder import *
from litex.soc.cores.led import LedChaser

# CRG ----------------------------------------------------------------------------------------------

class _CRG(LiteXModule):
    def __init__(self, platform, sys_clk_freq):
        self.rst    = Signal()
        self.cd_sys = ClockDomain()

        self.pll = pll = S7PLL(speedgrade=-2)
        self.comb += pll.reset.eq(self.rst)
        pll.register_clkin(platform.request("clk200"), 200e6)
        pll.create_clkout(self.cd_sys, sys_clk_freq)
        platform.add_false_path_constraints(self.cd_sys.clk, pll.clkin) # Ignore sys_clk to pll.clkin path created by SoC's rst.
        
# BaseSoC ------------------------------------------------------------------------------------------

class BaseSoC(SoCCore):
    def __init__(self, sys_clk_freq=200e6, with_led_chaser=True, **kwargs):
        platform = alinx_ax7203.Platform()

        # CRG --------------------------------------------------------------------------------------
        self.crg = _CRG(platform, sys_clk_freq)

        # SoCCore ----------------------------------------------------------------------------------
        #if kwargs["uart_name"] == "serial": kwargs["uart_name"] = "usb_uart" # Use USB-UART Pmod on JB.
        kwargs["uart_name"] = "serial"
        SoCCore.__init__(self, platform, sys_clk_freq, ident="LiteX SoC on Alinx AX7203", **kwargs)
        
        # Leds -------------------------------------------------------------------------------------
        if with_led_chaser:
            self.leds = LedChaser(
                pads         = platform.request_all("user_led"),
                sys_clk_freq = sys_clk_freq)

# Build --------------------------------------------------------------------------------------------

def main():
    from litex.build.parser import LiteXArgumentParser
    parser = LiteXArgumentParser(platform=alinx_ax7203.Platform, description="LiteX SoC on Artix7.")
    parser.add_target_argument("--sys-clk-freq", default=200e6, type=float, help="System clock frequency.")
    args = parser.parse_args()

    soc = BaseSoC(
        sys_clk_freq = args.sys_clk_freq,
        **parser.soc_argdict
    )
    builder = Builder(soc, **parser.builder_argdict)
    if args.build:
        builder.build(**parser.toolchain_argdict)

    if args.load:
        prog = soc.platform.create_programmer()
        prog.load_bitstream(builder.get_bitstream_filename(mode="sram"), device=1)

if __name__ == "__main__":
    main()

```

Help would be very appreciated.
Thank you in advance.

tarikh-kv, 2025-02-04:
The problem was I wrote `2e9/100e6`, instead of `1e9/200e6` in the platform file...

The build works now without a problem.

TarikHamedovic, 2025-02-04:
The problem was I wrote `2e9/100e6` instead of `1e9/200e6` in the platform file...

The build works now without a problem.
And the issue can be closed.

enjoy-digital, 2025-02-07:
Good @TarikHamedovic if you managed to understand/fix your issue. Feel free to contribute this new platform/target to LiteX Boards :)

closed 2025-02-07 by enjoy-digital