enjoy-digital/litex #2178: Converting Wishbone to riscv-steel custom bus

BMorgan1296, 2025-02-07:
Hi,

I am trying to incorporate the riscv-steel (https://github.com/riscv-steel/riscv-steel) processor into litex.
I have been able to get this working using a custom bus to Wishbone converter in simulation.

However, when I try to implement on FPGA, the processor does not boot properly. 
I have tried several various debugging methods for the past week to no avail.
I have attempted to use LiteScope to read in the values of the bus at boot, however as this is locked to the "sys" clock domain, I am unable to read these values at boot, as the scope it self also resets... I can read them in the middle of broken execution however I am finding it difficult to debug this.

Unfortunately, the riscv-steel repo does not document how this bus interface is truly meant to work. I have attempted to use the OBI2Wishbone used in the CV32E40P as a guide. Does anyone have any tips as to what I can do to debug this?

My implementation of the converter is below. 

```Python
steel_layout = [
    ("rw_address", 32),
    ("read_data", 32),
    ("read_request", 1),
    ("read_response", 1),
    ("write_data", 32),
    ("write_strobe", 4),
    ("write_request", 1),
    ("write_response", 1),
]

class SteelCore2Wishbone(LiteXModule):
    def __init__(self, steel_idbus, idbus):
        self.submodules.fsm = fsm = FSM(reset_state="IDLE")

        fsm.act("IDLE",
            If(steel_idbus.read_request,
                idbus.adr.eq(steel_idbus.rw_address),
                idbus.stb.eq(1),
                idbus.cyc.eq(1),
                idbus.we.eq(0),
                NextState("READ")
            ).Elif(steel_idbus.write_request,
                idbus.adr.eq(steel_idbus.rw_address),
                idbus.dat_w.eq(steel_idbus.write_data),
                idbus.stb.eq(1),
                idbus.cyc.eq(1),
                idbus.sel.eq(steel_idbus.write_strobe),
                idbus.we.eq(1),
                NextState("WRITE")
            ),
        )

        fsm.act("READ",
            If(idbus.ack,
                steel_idbus.read_data.eq(idbus.dat_r),
                steel_idbus.read_response.eq(1),
                NextState("IDLE")
            )
        )

        fsm.act("WRITE",
            If(idbus.ack,
                steel_idbus.write_response.eq(1),
                NextState("IDLE")
            )
        )
```

Any tips would be greatly appreciated. I'll add a waveform shortly to provide additional context to show what differs between simulation and FPGA.

enjoy-digital, 2025-02-18:
Hi @BMorgan1296,

thanks for adding this CPU to LiteX. It's already a good point if the CPU is able to boot in simulation. Would you be able to create a PR to integrate to integrate the CPU in LiteX, know to work with `litex_sim` and we could then help your figure out the limitations/fixes to do to run it on hardware. It's also possible it will work on some FPGA and not on others and we'll do test and help understanding the real issue/limitation.

BMorgan1296, 2025-02-20:
Have opened PR #2190 :)