enjoy-digital/litex #2182: Compiling software for 32-bits NaxRiscv SoC fails when forcing xlen=32 through the CLI

benjaminh-13, 2025-02-14:
Hello @Dolu1990 ,

First, thanks for your work on the NaxRiscv project.

When I try to compile software for a NaxRiscv SoC using a 32-bits version of the RISC-V toolchain, it fails as the toolchain cannot be detected.

I believe that this issue does not depend on my setup but originates from line 141 in NaxRiscv's ```core.py``` file :
https://github.com/enjoy-digital/litex/blob/bc1f1f52b0202baa52a24b869705bc71cfcb69d0/litex/soc/cores/cpu/naxriscv/core.py#L137-L142

as when manually passing argument ```--xlen=32``` to the litex CLI, attribute ```NaxRiscv.gcc_triple``` is forced to ```CPU_GCC_TRIPLE_RISCV64``` no matter ```xlen``` is 32 or 64, which prevents litex from finding any 32-bits RISC-V toolchain (I mean toolchains with prefix riscv32*).

Therefore, is there a plan to have ```core.py``` adapt attribute ```gcc_triple``` based on the xlen ?

For instance with such a condition
```python
NaxRiscv.gcc_triple           = CPU_GCC_TRIPLE_RISCV64 if xlen == 64 else CPU_GCC_TRIPLE_RISCV32
```

Actually, I noticed that by default, ```CPU_GCC_TRIPLE_RISCV32``` contains ```CPU_GCC_TRIPLE_RISCV64```.  (see below snippet from ```litex/soc/cores/cpu/__init__.py```)
https://github.com/enjoy-digital/litex/blob/bc1f1f52b0202baa52a24b869705bc71cfcb69d0/litex/soc/cores/cpu/__init__.py#L81-L91
Thus, could not line 141 be deleted to have ```NaxRiscv.gcc_triple``` default to  ```CPU_GCC_TRIPLE_RISCV32``` (using line 40), which would allow to cover 32-bits and 64-bits toolchain in any case ?
https://github.com/enjoy-digital/litex/blob/bc1f1f52b0202baa52a24b869705bc71cfcb69d0/litex/soc/cores/cpu/naxriscv/core.py#L40

Regards,

Dolu1990, 2025-02-17:
Hi,

> NaxRiscv.gcc_triple           = CPU_GCC_TRIPLE_RISCV64 if xlen == 64 else CPU_GCC_TRIPLE_RISCV32

Look good to me :)

> Thus, could not line 141 be deleted to have NaxRiscv.gcc_triple default to CPU_GCC_TRIPLE_RISCV32 (using line 40), which would allow to cover 32-bits and 64-bits toolchain in any case ?

I realy don't know if that would break something else in the litex flow.

i would say "NaxRiscv.gcc_triple           = CPU_GCC_TRIPLE_RISCV64 if xlen == 64 else CPU_GCC_TRIPLE_RISCV32" is the safest.

Do you wanna open a PR ?

benjaminh-13, 2025-02-17:
Hello,

Thanks for your quick response.

>> Thus, could not line 141 be deleted to have NaxRiscv.gcc_triple default to CPU_GCC_TRIPLE_RISCV32 (using line 40), which would allow to cover 32-bits and 64-bits toolchain in any case ?

>I realy don't know if that would break something else in the litex flow.

Ok, I completely understand.

Yes, I would like to open a PR if possible.

Thanks,
Regards,

Dolu1990, 2025-02-17:
> Yes, I would like to open a PR if possible.

Sure ^^

benjaminh-13, 2025-02-17:
As discussed, here is the PR #2187

enjoy-digital, 2025-02-18:
@benjaminh-13, this is merged, thanks!

benjaminh-13, 2025-02-18:
Thank you @Dolu1990 and @enjoy-digital, I am now closing the issue.

closed 2025-02-18 by benjaminh-13

Dolu1990, 2025-02-18:
Thanks for the PR ^^