enjoy-digital/litex #2195: Cannot install Migen on a new install

pcotret, 2025-03-04:
- OS Debian 12.
- I wanted to install LiteX on a new machine.

```bash
$ ./litex_setup.py --init --install --user --config=full
[...]
[   0.129] Installing migen Git repository...
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.
```

pcotret, 2025-03-04:
Maybe it may be easier to do the install inside a virtual environment?

pepijndevos, 2025-03-10:
Yes definitely use a virtual environment if your OS manages the environment

pcotret, 2025-03-10:
OK, I close the issue.

closed 2025-03-10 by pcotret

trabucayre, 2025-03-10:
The solution I use is to create the file `$HOME/.config/pip/pip.conf` with this content:
```
[global]
break-system-packages = true
```

pcotret, 2025-03-10:
> The solution I use is to create the file `$HOME/.config/pip/pip.conf` with this content:
> 
> ```
> [global]
> break-system-packages = true
> ```

Well, I don't want to "break" everything, I'll stay in my virtual environment 😉