enjoy-digital/litex #2200: Litex SoC binds to non-existent ports when generating a NaxRiscv-based SoC with --with-jtag-instruction

benjaminh-13, 2025-03-05:
Hello,

# Issue
I'm trying to implement a NaxRiscv-based SoC on FPGA (Digilent Genesys 2) but synthesis fails, as the SoC's top level module instantiating the Nax tries to connect to port which do not exist.

Precisely, I'm trying to add a JTAG debug module within the Nax, which thus exposes I/O ports in the Nax's top level. For some reason, when integrating the Nax in the SoC's top level, Litex seems to detect different signal names for the JTAG ports, than the ones actually exposed by the Nax, which thus causes an error during synthesis.

Instantation of the NaxRiscv in the SoC top module (namely the verilog file ```build/gateware/digilent_genesys2.v```)
```verilog
//------------------------------------------------------------------------------
// Instance NaxRiscvLitex_ac66cf5e80a0da4a2f58fc5f9f240934 of NaxRiscvLitex_ac66cf5e80a0da4a2f58fc5f9f240934 Module.
//------------------------------------------------------------------------------
NaxRiscvLitex_ac66cf5e80a0da4a2f58fc5f9f240934 NaxRiscvLitex_ac66cf5e80a0da4a2f58fc5f9f240934(
        /* Remaining ports connections */
        .jtag_instruction_capture           (basesoc_jtag_capture),
        .jtag_instruction_clk               (basesoc_jtag_clk),
        .jtag_instruction_enable            (basesoc_jtag_enable),
        .jtag_instruction_reset             (basesoc_jtag_reset),
        .jtag_instruction_shift             (basesoc_jtag_shift),
        .jtag_instruction_tdi               (basesoc_jtag_tdi),
        .jtag_instruction_update            (basesoc_jtag_update),
        /* Remaining ports connections */
       	);
```

Definition of the NaxRiscv top module
```verilog
module NaxRiscvLitex_ac66cf5e80a0da4a2f58fc5f9f240934 (
	/* Remaining ports declaration */
	input  wire          jtag_instruction_instruction_tdi,
	input  wire          jtag_instruction_instruction_enable,
	input  wire          jtag_instruction_instruction_capture,
	input  wire          jtag_instruction_instruction_shift,
	input  wire          jtag_instruction_instruction_update,
	input  wire          jtag_instruction_instruction_reset,
	output wire          jtag_instruction_instruction_tdo,
	input  wire          jtag_instruction_clk
	/* Remaining ports declaration */
);
	/* RTL */
endmodule
```

As one can see from the above RTL snippets, except for ```jtag_instruction_clk``` which is correct, JTAG signals are declared with prefix ```jtag_instruction_instruction``` whilst the SoC is mapping to ports with prefix ```jtag_instruction```.

# Build options

Here are my build options to try and recreate the issue :
```bash
python3 -m litex_boards.targets.digilent_genesys2.py \
--cpu-type=naxriscv \
--toolchain=vivado \
--build \
--bios-console=lite --bios-format=double \
--synth-mode=vivado \
--bus-standard=wishbone --bus-data-width=32 --bus-address-width=32 --bus-interconnect=crossbar \
--watchdog-width=32 \
--uart-name=serial --uart-baudrate=921600 --uart-fifo-depth=16 \
--l2-size=1024 --integrated-rom-size=131072 --integrated-sram-size=16536 --integrated-main-ram-size=0 \
--sys-clk-freq=200000000.0 \
--no-netlist-cache --update-repo=latest \
--cpu-variant=standard --xlen=32 --cpu-count=1 \
--l2-bytes=131072 --l2-ways=8 --with-rvc --with-fpu --with-coherent-dma \
--with-jtag-instruction
```

Actually, I use a modified version of ```litex_boards.targets.digilent_genesys2``` in which I added  the following lines (right-after the ```BaseSoc``` instantiation in ```main()```) :

```python
from litex.soc.cores.jtag import XilinxJTAG
soc.jtag = jtag = XilinxJTAG(XilinxJTAG.get_primitive(soc.platform.device), chain=4)
soc.comb += [
    soc.cpu.jtag_reset.eq(jtag.reset),
    soc.cpu.jtag_capture.eq(jtag.capture),
    soc.cpu.jtag_shift.eq(jtag.shift),
    soc.cpu.jtag_update.eq(jtag.update),
    soc.cpu.jtag_clk.eq(jtag.tck),
    soc.cpu.jtag_tdi.eq(jtag.tdi),
    soc.cpu.jtag_enable.eq(True),
    jtag.tdo.eq(soc.cpu.jtag_tdo),
]
```

Note: I added the latter modifications following this tutorial from Litex Wiki [JTAG GDB Debugging with VexRiscv SMP NaxRiscv VexiiRiscv CPUs](https://github.com/enjoy-digital/litex/wiki/JTAG-GDB-Debugging-with-VexRiscv-SMP-NaxRiscv-VexiiRiscv-CPUs#debugging-on-physical-hardware-directly-though-bscane2-jtag-primitive), as I would like to route the JTAG module to the FPGA's built-in JTAG.

# Question

I was wondering if anyone knows why Litex catches wrong names specifically for these signals ? Also, perhaps, is there a way to force the name of these ```XilinxJTAG``` signals to have prefix ```jtag_instruction``` instead of ```jtag_instruction_instruction``` ?

Thanks,
Regards,

atkarim, 2025-03-06:
Hello @benjaminh-13,

I suggest 2 solutions: 

- either make a modification on the NaxRiscv side, 
- or make a modification on the Litex side.

#### On the NaxRiscv side

You'll need to modify the file that generates the SoC for Litex, NaxSoc, by changing this line :

https://github.com/SpinalHDL/NaxRiscv/blob/43716b5770e8ac689e9778056c061da577dcf17e/src/main/scala/naxriscv/platform/litex/NaxSoc.scala#L209

by this one : 

```scala
if (withJtagInstruction) debug.instruction.setName("jtag")
```

What happens is that the string `jtag_instruction` is added to the variable name `instruction` when the verilog is generated.
There must be a way to ignore the variable name during generation, but I haven't found it yet.

#### On the Litex side

You need to modify the core.py file associated with NaxRiscv.

https://github.com/enjoy-digital/litex/blob/48e65c199de6bd1d5aead61d7f515356cc7ecab8/litex/soc/cores/cpu/naxriscv/core.py#L428-L437

You need to update the assignments with the “jtag_instruction_instruction” prefix.

```python
self.cpu_params.update(
    i_jtag_instruction_clk     = self.jtag_clk,
    i_jtag_instruction_instruction_enable  = self.jtag_enable,
    i_jtag_instruction_instruction_capture = self.jtag_capture,
    i_jtag_instruction_instruction_shift   = self.jtag_shift,
    i_jtag_instruction_instruction_update  = self.jtag_update,
    i_jtag_instruction_instruction_reset   = self.jtag_reset,
    i_jtag_instruction_instruction_tdi     = self.jtag_tdi,
    o_jtag_instruction_instruction_tdo     = self.jtag_tdo,
)
```

After one of these two modifications, the Vivado synthesis should work.

Best regards

benjaminh-13, 2025-03-06:
Hello @atkarim,

Thank you for the very quick response.

I tried solution 2, namely bringing changes to the NaxRiscv's ```core.py``` in Litex and now synthesis succeeds. As a consequence, I am closing this issue.

Thanks again,
Best regards,

PR #2210 (2025-03-24): PR following enjoy-digital#2200 closure
Hello,

I found in the now resolved #2200 that the mapping of some JTAG instruction ports in [litex/litex/soc/cores/cpu/naxriscv/core.py](http://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/naxriscv/core.py#L428) are not matching the name of the actual NaxRiscv's JTAG instruction ports.

Following the discussion in #2200, here is how it could be adressed.

Best regards,
closed 2025-03-06 by benjaminh-13