enjoy-digital/litex #2201: SD-mode failing on LFE5U-25F boards

machdyne, 2025-03-07:
Hello. I'm trying to debug an issue that I see on some boards where spisdcard works but sdcard (sd-mode) fails. There are external 20K pull-ups on all sdcard signals. I have tried different microsd card brands.

This seems to only affect our LFE5U-12/25F boards, but not LFE5U-45F (it does work on these), maybe that provides some hint as to the problem. Timing passes on both.

Any ideas?

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (b622ebda)

 LiteX git sha1: 2bcbbafdd

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 48MHz
BUS:            wishbone 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64.0KiB
SRAM:           6.0KiB
SDRAM:          512.0MiB 16-bit @ 192MT/s (CL-6 CWL-5)
MAIN-RAM:       512.0MiB

--========== Initialization ============--

Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |01110000| delays: 02+-01
  m0, b01: |00000000| delays: -
  m0, b02: |00000000| delays: -
  m0, b03: |00000000| delays: -
  best: m0, b00 delays: 02+-01
  m1, b00: |01110000| delays: 02+-01
  m1, b01: |00000000| delays: -
  m1, b02: |00000000| delays: -
  m1, b03: |00000000| delays: -
  best: m1, b00 delays: 02+-01
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 57.9MiB/s
   Read speed: 31.1MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from SDCard in SD-Mode...
Booting from boot.json...
Setting SDCard clk freq to 375 KHz
CMD0: GO_IDLE
cmdevt: wait for event & 0x1
cmdevt: 00000001
00000000 00000000 00000000 00000000
CMD8: SEND_EXT_CSD, arg: 0x000001aa
cmdevt: wait for event & 0x1
cmdevt: 00000001
00000000 00000000 00000008 000001aa
Setting SDCard clk freq to 24 MHz
CMD55: APP_CMD
cmdevt: wait for event & 0x1
cmdevt: 00000001
00000000 00000800 0001aa3f 37000001
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
cmdevt: wait for event & 0x1
cmdevt: 00000001
00080000 01aa3f37 0000013f ffffffff
CMD2: ALL_SEND_CID
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CID Register: 0xffffffffffffffffffffffffffffffff
Manufacturer ID: 0xffff
Application ID 0xffff
Product name: �����
CRC: ff
Production date(m/yy): 15/255
PSN: ffffffff
OID: ��
CMD3: SET_RELATIVE_ADDRESS
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffff3f ffffffff
CMD10: SEND_CID
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CMD9: SEND_CSD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CSD Register: 0xffffffffffffffffffffffffffffffff
Max data transfer rate: 255 MB/s
Max read block length: 32768 bytes
Device size: 0 GB
CMD7: SELECT_CARD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffff3f ffffffff
CMD55: APP_CMD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffff3fff ffffff3f ffffffff
ACMD6: SET_BUS_WIDTH
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff3f ffffffff
CMD6: SWITCH_FUNC
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff3f ffffffff
dataevt: wait for event & 0x1
dataevt: 00000005
Booting from boot.bin...
Setting SDCard clk freq to 375 KHz
CMD0: GO_IDLE
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff3f ffffffff
CMD8: SEND_EXT_CSD, arg: 0x000001aa
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff08 000001aa
Setting SDCard clk freq to 24 MHz
CMD55: APP_CMD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff0800 0001aa3f 37000001
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff080000 01aa3f37 0000013f ffffffff
CMD2: ALL_SEND_CID
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CID Register: 0xffffffffffffffffffffffffffffffff
Manufacturer ID: 0xffff
Application ID 0xffff
Product name: �����
CRC: ff
Production date(m/yy): 15/255
PSN: ffffffff
OID: ��
CMD3: SET_RELATIVE_ADDRESS
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffff3f ffffffff
CMD10: SEND_CID
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CMD9: SEND_CSD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffffff ffffffff
CSD Register: 0xffffffffffffffffffffffffffffffff
Max data transfer rate: 255 MB/s
Max read block length: 32768 bytes
Device size: 0 GB
CMD7: SELECT_CARD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffffffff ffffff3f ffffffff
CMD55: APP_CMD
cmdevt: wait for event & 0x1
cmdevt: 00000001
ffffffff ffff3fff ffffff3f ffffffff
ACMD6: SET_BUS_WIDTH
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff3f ffffffff
CMD6: SWITCH_FUNC
cmdevt: wait for event & 0x1
cmdevt: 00000001
ff3fffff ffff3fff ffffff3f ffffffff
dataevt: wait for event & 0x1
dataevt: 00000005
SDCard boot failed.
No boot medium found
```

gatecat, 2025-09-05:
Reducing the SD clock frequency (`SDCARD_CLK_FREQ`) in the BIOS to 5MHz seems to make it work, too

gatecat, 2025-09-05:
Aha: https://github.com/enjoy-digital/litex/pull/2326

The problem is that SDR tristates weren't using a dedicated IO flipflop for the tristate part, so random variations in routing were causing an extra delay in the tristate compared to the other signals (the reason it was less reliable with DDR is because that makes the overall design bigger and therefore the routing longer).

This now works reliably (including booting Linux) for me on the Trellisboard at the normal speed.

PR #2326 (2025-09-05): build/lattice: Add SDR tristate specialisation for ECP5
Required for reliable litesdcard on ECP5 in SD mode; fixes #2201
closed 2025-09-05 by trabucayre