enjoy-digital/litex #2202: Vexiiriscv doesn't simulate

pepijndevos, 2025-03-10:
I just completely updated my installation and ran a basic command as a sanity check. Serv and vexriscv work but vexiiriscv doesn't:

```
$ litex_sim --cpu-type=vexiiriscv
```

And then I get a build error:

```
make: Entering directory '/home/pepijn/code/apicula/build/sim/gateware'
mkdir -p modules
make -C modules -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/Makefile
make[1]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules'
mkdir -p xgmii_ethernet
make MOD=xgmii_ethernet -C xgmii_ethernet -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/xgmii_ethernet/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/xgmii_ethernet'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/xgmii_ethernet/../.. -o xgmii_ethernet.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/xgmii_ethernet/xgmii_ethernet.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o tapcfg.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/tapcfg.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o taplog.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/taplog.c
cc -levent -shared -fPIC -lz -Wl,-soname,xgmii_ethernet.so -o xgmii_ethernet.so xgmii_ethernet.o tapcfg.o taplog.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/xgmii_ethernet'
cp xgmii_ethernet/xgmii_ethernet.so xgmii_ethernet.so
mkdir -p ethernet
make MOD=ethernet -C ethernet -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/ethernet/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/ethernet'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/ethernet/../.. -o ethernet.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/ethernet/ethernet.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o tapcfg.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/tapcfg.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o taplog.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/taplog.c
cc -levent -shared -fPIC -Wl,-soname,ethernet.so -o ethernet.so ethernet.o tapcfg.o taplog.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/ethernet'
cp ethernet/ethernet.so ethernet.so
mkdir -p serial2console
make MOD=serial2console -C serial2console -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2console/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/serial2console'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2console/../.. -o serial2console.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2console/serial2console.c
cc -levent -shared -fPIC -Wl,-soname,serial2console.so -o serial2console.so serial2console.o
rm serial2console.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/serial2console'
cp serial2console/serial2console.so serial2console.so
mkdir -p serial2tcp
make MOD=serial2tcp -C serial2tcp -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2tcp/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/serial2tcp'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2tcp/../.. -o serial2tcp.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/serial2tcp/serial2tcp.c
cc -levent -shared -fPIC -Wl,-soname,serial2tcp.so -o serial2tcp.so serial2tcp.o
rm serial2tcp.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/serial2tcp'
cp serial2tcp/serial2tcp.so serial2tcp.so
mkdir -p clocker
make MOD=clocker -C clocker -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/clocker/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/clocker'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/clocker/../.. -o clocker.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/clocker/clocker.c
cc -levent -shared -fPIC -Wl,-soname,clocker.so -o clocker.so clocker.o
rm clocker.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/clocker'
cp clocker/clocker.so clocker.so
mkdir -p spdeeprom
make MOD=spdeeprom -C spdeeprom -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/spdeeprom/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/spdeeprom'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/spdeeprom/../.. -o spdeeprom.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/spdeeprom/spdeeprom.c
cc -levent -shared -fPIC -Wl,-soname,spdeeprom.so -o spdeeprom.so spdeeprom.o
rm spdeeprom.o
make[2]: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware/modules/spdeeprom'
cp spdeeprom/spdeeprom.so spdeeprom.so
mkdir -p gmii_ethernet
make MOD=gmii_ethernet -C gmii_ethernet -f /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/gmii_ethernet/Makefile
make[2]: Entering directory '/home/pepijn/code/apicula/build/sim/gateware/modules/gmii_ethernet'
cc -c -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -I/home/pepijn/code/apicula/litex/litex/build/sim/core/modules/gmii_ethernet/../.. -o gmii_ethernet.o /home/pepijn/code/apicula/litex/litex/build/sim/core/modules/gmii_ethernet/gmii_ethernet.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o tapcfg.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/tapcfg.c
cc -Wall -O3 -ggdb -fPIC -Werror -I/home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/include -c -o taplog.o /home/pepijn/code/apicula/pythondata-misc-tapcfg/pythondata_misc_tapcfg/data/src/lib/taplog.c
cc -levent -shared -fPIC -lz -Wl,-soname,gmii_ethernet.so -o gmii_ethernet.so gmii_ethernet.o tapcfg.o taplog.o
[... 41 lines cut ...]
	-Wno-BLKANDNBLK \
	-Wno-WIDTH \
	-Wno-COMBDLY \
	-Wno-CASEINCOMPLETE \
	--relative-includes
%Error-PINNOTFOUND: /home/pepijn/code/apicula/pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/VexiiRiscvLitex_e92c04213ef0ab400048eee18d35deed.v:11554:6: Pin not found: 'rd_dataEn'
                                                                                                                                                                    : ... Suggested alternative: 'rd_data'
11554 |     .rd_dataEn (1'b1                                                ),  
      |      ^~~~~~~~~
                    ... For error description see https://verilator.org/warn/PINNOTFOUND?v=5.032
%Error-PINNOTFOUND: /home/pepijn/code/apicula/pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/VexiiRiscvLitex_e92c04213ef0ab400048eee18d35deed.v:11544:6: Parameter not found: 'rdLatency'
11544 |     .rdLatency      (1         )
      |      ^~~~~~~~~
%Error-PINNOTFOUND: /home/pepijn/code/apicula/pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/VexiiRiscvLitex_e92c04213ef0ab400048eee18d35deed.v:11579:6: Pin not found: 'rd_dataEn'
                                                                                                                                                                    : ... Suggested alternative: 'rd_data'
11579 |     .rd_dataEn (1'b1                                                ),  
      |      ^~~~~~~~~
%Error-PINNOTFOUND: /home/pepijn/code/apicula/pythondata-cpu-vexiiriscv/pythondata_cpu_vexiiriscv/verilog/VexiiRiscvLitex_e92c04213ef0ab400048eee18d35deed.v:11569:6: Parameter not found: 'rdLatency'
11569 |     .rdLatency      (1         )
      |      ^~~~~~~~~
%Error: Exiting due to 4 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make: *** [/home/pepijn/code/apicula/litex/litex/build/sim/core/Makefile:44: sim] Error 1
make: Leaving directory '/home/pepijn/code/apicula/build/sim/gateware'
```

Dolu1990, 2025-03-10:
Hi,
I had some unpushed local changes.
If you pull the vexiiriscv litex python data it should be good.
Thanks :D

closed 2025-03-10 by Dolu1990

pepijndevos, 2025-03-10:
I have updated but now it seems to hang in the simulation

```
[serial2tcp] loaded (0x6146cfa0c2f0)
[spdeeprom] loaded (addr = 0x0)
[clocker] loaded
[gmii_ethernet] loaded (0x6146cfa0c2f0)
[jtagremote] loaded (0x6146cfa0c2f0)
[xgmii_ethernet] loaded (0x6146cfa0c2f0)
[serial2console] loaded (0x6146cfa0c2f0)
[ethernet] loaded (0x6146cfa0c2f0)
[clocker] sys_clk: freq_hz=1000000, phase_deg=0
```
[vsim just sits there at 100% CPU]

pepijndevos, 2025-03-10:
nvm it just took much longer than I've ever seen before with other cores to get to the litex prompt.

Dolu1990, 2025-03-10:
Ahhhh right ^^
Fixed with https://github.com/enjoy-digital/litex/pull/2203

Thanks :)

commit 469a07d0 (2025-03-17): Merge pull request #2203 from enjoy-digital/Dolu1990-patch-4

cores/cpu/vexiiriscv : Fix #2202 long boot time
commit 92a64e42 (2025-03-17): cores/cpu/vexiiriscv : Fix #2202 long boot time