enjoy-digital/litex #2208: C program Tang Primer 20K

russo00, 2025-03-14:
Hello everyone! 
I recently started working with LiteX, and I have a Tang Primer 20K board. I successfully flashed LiteX onto it, and on boot, I see the LiteX welcome message.
Now, I want to compile a simple C program and upload it via UART using the command:
litex_term --kernel=firmware.bin /dev/ttyUSB0
I wrote a basic C program along with a Makefile and compiled it using:
riscv32-unknown-elf-gcc -O2 -nostdlib -ffreestanding -o firmware.elf main.c
However, I got this warning:
/home/lan/riscv32/lib/gcc/riscv32-unknown-elf/14.2.0/../../../../riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 00010094
The firmware.elf file was generated, but when I try to upload it with:
litex_term --kernel=firmware.bin /dev/ttyUSB0
Nothing happens—the command just hangs.
Additionally, I’m struggling to find example C code for LiteX, such as how to interface with UART, timers, or GPIO.
Can anyone help me with these issues?

main.c

#include <stdio.h>
#include <stdlib.h>
int main(void) {
//printf("Hello, LiteX!\n");
while (1) {
//printf("Running...\n");
for (volatile int i = 0; i < 1000000; i++);
}
return 0;
}

Makefile:

TARGET = firmware
SRC = main.c
OBJS = $(SRC:.c=.o)
CC = riscv32-unknown-elf-gcc
CFLAGS = -march=rv32im -mabi=ilp32 -Os -ffreestanding -nostdlib -Wall
LDFLAGS = -T libc/picolibc.ld -nostdlib
all: $(TARGET).bin
$(TARGET).elf: $(OBJS)
$(CC) $(CFLAGS) $(LDFLAGS) -o $@ $\^
%.bin: %.elf
riscv32-unknown-elf-objcopy -O binary $< $@
clean:
rm -f $(OBJS) $(TARGET).elf $(TARGET).bin

russo00, 2025-03-15:
Thank you, kind person! I ran the command, and the files were created in the demo folder.

Now I’m not sure how to correctly upload the demo.elf file to the board via UART.

[image]

trabucayre, 2025-03-15:
It's not the `.hex` but the `.bin` you have to uses:
```bash
litex_term /dev/ttyUSBx --kernel demo.bin
```

Since you uses a tangPrimer 20k with an **FT2232** you have to replaces `/dev/ttyUSBx` by `/dev/ttyUSB1` (assuming you don't have more USB<->UART adapters connected to your PC).

russo00, 2025-03-15:
I run the command:
litex_term /dev/ttyUSB0 --kernel demo.bin
or
litex_term /dev/ttyUSB1 --kernel demo.bin

But there is no response—the command just hangs. I don’t know what to do.

russo00, 2025-03-29:
[image]

Here’s the English translation of your text:

"I've made some progress. Earlier, I ran the command 'litex_bare_metal_demo --build-path build/sipeed_tang_primer_20k' in the targets folder, and compilation files appeared in the demo folder. Then, in the demo folder, I run 'litex_term /dev/ttyUSB1 --kernel demo.bin' and press Enter, and the LiteX prompt appears. Next, I type 'serialboot', press Enter, and the board flashing process starts, but it stops. I’ve attached a photo. I don't understand why the flashing process stops."

If you need any corrections or refinements, let me know!

alberto-grl, 2025-03-30:
I saw errors like yours while using the Sipeed supplied USB cable. Try another cable.

Or maybe it could be a problem with the braille reader, active by default, that sometimes kicks in
https://www.reddit.com/r/digirig/comments/v4rxer/fix_for_ubuntu_2204_serial_device_issue_credit/
Errors would be different, but in doubt uninstall it.

Regards

trabucayre, 2025-04-28:
@alberto-grl it's maybe the reason yes.
I have re-tested the full procedure and not able to reproduce this issue:
```bash
# Cleanup
rm -rf build demo demo.bin demo.fbi
# For Tang Primer 20k with dock standard
./sipeed_tang_primer_20k.py --build --load
# Or for Tang Primer 20k with dock lite
./sipeed_tang_primer_20k.py --dock lite --build --load
# Prepare demo.bin
litex_bare_metal_demo --build-path build/sipeed_tang_primer_20k
# load
litex_term --kernel demo.bin /dev/ttyUSB1
```

alberto-grl, 2025-05-01:
Now I have a fairly stable setup. For some reason I have 100% success with the Gowin programmer, OpenFPGA loader sometimes looses communication. Don't know if changing speed or some other parameter would help.
Litex_term works ok, but always at first connection does not show output from the board. Close it, restart and i see replies to input I previously gave. Than it works OK.

trabucayre, 2025-05-05:
> Now I have a fairly stable setup. For some reason I have 100% success with the Gowin programmer, OpenFPGA loader sometimes looses communication. Don't know if changing speed or some other parameter would help. 

FTDI clone based on bl702 is not always stable and contains some hooks for gowin programmer. Sometime changing USB cable or removing USB hub improves communication.

> Litex_term works ok, but always at first connection does not show output from the board. Close it, restart and i see replies to input I previously gave. Than it works OK. Same thing with Putty or minicom.

Yes: it's a known issue also related to the bl702 firmware.