enjoy-digital/litex #2212: Microwatt fails to build

nyaasen, 2025-03-23:
Hi All,

Microwatt appears to fail to build. I have tried both the ULX3S and the Arty A7. It looks like there's an issue with microwatt at the VHD2V converter section. I went ahead and tried with neorv32 and that was able to go through just fine, so I do not believe it is an issue with VHD2V itself. Any suggestions on how to debug this? 

Thanks

```C
INFO:SoC:Controller ctrl added.
Traceback (most recent call last):
  File "/home/[user]/litex/./litex-boards/litex_boards/targets/radiona_ulx3s.py", line 193, in <module>
    main()
    ~~~~^^
  File "/home/[user]/litex/./litex-boards/litex_boards/targets/radiona_ulx3s.py", line 166, in main
    soc = BaseSoC(
        device                 = args.device,
    ...<7 lines>...
        with_spi_flash         = args.with_spi_flash,
        **parser.soc_argdict)
  File "/home/[user]/litex/./litex-boards/litex_boards/targets/radiona_ulx3s.py", line 105, in __init__
    SoCCore.__init__(self, platform, sys_clk_freq, ident="LiteX SoC on ULX3S", **kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[user]/litex/litex/litex/soc/integration/soc_core.py", line 218, in __init__
    self.add_cpu(
    ~~~~~~~~~~~~^
        name          = str(cpu_type),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        variant       = "standard" if cpu_variant is None else cpu_variant,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        reset_address = None if integrated_rom_size else cpu_reset_address,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        cfu           = cpu_cfu)
        ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[user]/litex/litex/litex/soc/integration/soc.py", line 1197, in add_cpu
    self.cpu = cpu_cls(self.platform, variant)
               ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/[user]/litex/litex/litex/soc/cores/cpu/microwatt/core.py", line 135, in __init__
    self.cpu_vhd2v_converter = VHD2VConverter(platform,
                               ~~~~~~~~~~~~~~^^^^^^^^^^
        top_entity    = "microwatt_wrapper",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        build_dir     = os.path.abspath(os.path.dirname(__file__)),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        force_convert = ("ghdl" in self.variant),
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/[user]/litex/litex/litex/build/vhd2v_converter.py", line 74, in __init__
    assert (self._params is None) ^ (self._instance is None)
           ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
AssertionError```

trabucayre, 2025-04-25:
Hi and sorry for the delay.
Fixed with this commit https://github.com/enjoy-digital/litex/commit/275a1681c64bd8188a61e97564e0cb5852d79f51
Thanks

nyaasen, 2025-05-03:
Thank you, and please do not worry about the delay. You guys do a great job.

This fixes the issue in VHD2V. I am able to get to the synthesis portion. 

Thank you very much!

closed 2025-05-03 by nyaasen