enjoy-digital/litex #2280: yosys v > 0.48 causes build error

DaveBerkeley, 2025-07-10:
I'm seeing the error 

ERROR: Unable to place cell 'VexRiscv.dataCache_1', no BELs remaining to implement cell type '$scopeinfo'

when using versions of yosys > 0.48. I'm guessing that there is some config option that I'm missing.

I'm building a VexRiscv SoC for the ECP5 colorlight i9.

gsomlo, 2025-07-10:
Would you mind sharing the exact command line you've used to build
your bitstream?

DaveBerkeley, 2025-07-11:
```
cd ~/litex/litex-boards/litex_boards/targets
./colorlight_i5.py --build

Info: Placed 110 cells based on constraints.
ERROR: Unable to place cell 'VexRiscv.dataCache_1', no BELs remaining to implement cell type '$scopeinfo'
0 warnings, 1 error
```
I'm using  litex-boards 42d4615aace8bf9ee and litex 6c2beeec1df16edb48cbf11

fails with 

`Yosys 0.49 (git sha1 427b5a251, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)`

builds correctly with 

`Yosys 0.48 (git sha1 aaa534749, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)`

gsomlo, 2025-07-11:
On Fri, Jul 11, 2025 at 12:39:15AM -0700, DaveBerkeley wrote:
> cd ~/litex/litex-boards/litex_boards/targets
> ./colorlight_i5.py --build
> 
> ...
>
> Yosys 0.49 (git sha1 427b5a251, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)
> 
> builds correctly with
> 
> Yosys 0.48 (git sha1 aaa534749, g++ 13.3.0-6ubuntu2~24.04 -fPIC -O3)

I tried it (successfully) with yosys 0.55 (and a relatively recent
nextpnr -- commit 0ebd7af).

Maybe whatever issues there were in 0.49 were fixed in subsequent
versions?

DaveBerkeley, 2025-07-11:
I did a binary search on yosys builds.

It failed for me on v0.55 v0.54 v0.51 v0.50 v0.49 passed on v0.42 and below

I used a script to build yosys 

```
#!/bin/bash

cd /tmp
rm -rf yosys
git clone https://github.com/YosysHQ/yosys.git
cd yosys
git checkout $1
git submodule update --init --recursive

make config-gcc
make -j 12
sudo make install
```
Perhaps it is an issue with my nextpnr interacting with yosys? I used the same version for all 

`"nextpnr-ecp5" -- Next Generation Place and Route (Version 0.6-3build5)`